playfultechnology/audio-guestbook #1: "Not declared in this scope"

Jessman11289, 2022-06-07:
I am running into issues compiling for my teensyLC and get the two below errors. I am very novice so any assistance would be helpful, thank you!
[image]

embankment, 2022-06-29:
Jessman: Although the errors you are seeing do not point to an incompatibility between boards, the code was written for a Teensy 4.0 (4.1 might work with minor tweaks).  The Teensy LC has incomplete support for audio library functions and does not have the processing power to pull off this project.

DD4WH, 2022-08-04:
Please close this issue, if the question is answered