playfultechnology/audio-guestbook #4: fatal error: MTP_Teensy.h: No such file or directory / Question on the library used

SteffenGa, 2022-08-03:
When compiling, I get the error that MTP_Teensy.h could not be found. This library appears to be required with the latest additions from @DD4WH 

The problem I have is that I couldn't find much information on how to get this library, i.e. I cannot find any library named like that in the Teensyduino Library Manager, and I don't know from which developer this is from to be able to locate and install the correct one.

Thank you so much in advance for your helpful tip, and overall for this great project :-)

DD4WH, 2022-08-04:
Have a look here:
https://forum.pjrc.com/threads/70553-Teensy-4-0-based-Audio-Guestbook?p=310485&viewfull=1#post310485
Good luck with building !

DD4WH, 2022-08-04:
@SteffenGa 
Extensive documentation has been added to the readme (https://github.com/DD4WH/audio-guestbook)
If this answers your question, please close this issue.

SteffenGa, 2022-08-04:
That's great, thank you so much. Will close the issue.

closed 2022-08-04 by SteffenGa