playfultechnology/audio-guestbook #9: wait() function leads to unwanted modes in certain situations

SteffenGa, 2022-08-18:
I did some testing with my telephone (German W48 phone, contact closes on handheld lift), and discovered the following behaviour: When uplifting the handset and before the greeting.wav is played, there is a custom wait() function being called, with input parameter of 1000ms. 

`    case Mode::Prompting:
      // Wait a second for users to put the handset to their ear
      wait(1000);`

If I now hang up the handset during the wait (so before the 1000ms are over) the phone **stays** in mode PROMTING, and I can see the debug message "Button (pin 0) Release".

If I replace the wait(1000) with a delay(1000) in this line of code, everything works as expected, meaning, also a short uplifting of the handset (<1000ms) will not lead to an unwanted state and brings the phone finally into READY mode again. 

I don't really understand where this problem comes from (I think it is implementation related), and I'm not sure if it also affects other areas of the code where the wait() function is being used under certain situations.

alastaira, 2022-08-21:
You're correct - the custom wait() function is used instead of delay() specifically because it is non-blocking. So, any button changes that occur during that 1000mS period are "absorbed" but not acted upon. I'll make a note to check whether the handset has been replaced during the waittime and update the state accordingly.

kkidfall, 2022-08-21:
@alastaira that would be awesome! Is there a code we can use to do a max recording time to reset the system to waiting as well?

rchopik, 2022-09-21:
@alastaira Thanks for looking into this! Do you have any update on how this can be addressed? I found the same scenario occurs if someone hastily drops the receiver and it "bounces" on the phone body. It ends the recording, but the bounce triggers the pins and the greeting begins playing again while the receiver is not lifted. 

A max record time I think would be a viable alternative as well if the delay function can't absorb other button changes.

woozi22, 2022-12-04:
Hi @SteffenGa thanks for bringing this up. I also saw this behavior with my phone. That generated some quite long recordings and I wondered why.
@alastaira what is the recommended solution here? Change the wait command to delay? 
I would be happy to remove this issue since it seems that it is quite likely that people do that "bounce" when hanging up.

Thanks :)

woozi22, 2022-12-04:
What if I reduce the wait time to 1 ms (or delete that command at all) and instead of it I add one "blank" second to my greeting file. Shouldn't this be the same?

faleszumu, 2023-01-09:
hello, anyone solve the problem?

dlew8weld, 2023-03-11:
Instead of adding additional checks during the wait time I have added an additional check to the main loop to stop the recording, The .risingEdge() only returns true for one cycle and this is missed if the handset is replaced during the wait function. 

I added an OR condition to check if the handset is replaced and has been replaced for longer than 1.5seconds. I felt this was the lowest risk solution for me without diving too much into the code. 

   
```
// Handset is replaced
if(buttonRecord.risingEdge() || (buttonRecord.duration()>=1500 && buttonRecord.read())) {
  playWav1.stop();
  mode = Mode::Ready; print_mode();
  return;
}
```
If your recording is less than the 1.5seconds you would probably also need to add this check to the recording mode