playfultechnology/audio-guestbook #12: Crack noise while speaking

woozi22, 2022-09-14:
Hi all,

I really appreciate that you created that code & video to rebuild this fantastic idea. Unfortunately I am currently struggeling with one issue. Everything works fine except of some cracking noises when I speak. First I thought this is caused by the old microphone capsules but I changed it and it is still there. 
[00033.zip](https://github.com/playfultechnology/audio-guestbook/files/9564005/00033.zip)

I uploaded a test recording. In this case there are not many crack noises, sometimes it is worse. But after 13 seconds you can hear it.
And it only occurs when I am speaking. When I don't say anything it is all clear.

Here is a list what I already tried:
- Checked all soldering points
- Exchanged AUX cable to a double shielded
- Used a ground loop isolator
- Changed old microphone to a AOL5024 L HD R
- Increased the audio buffer from 60 to 125 (using a SanDisk 32 GB sd card with class 10 - should be fast enough)
- added some foam in front of the microphone cause I thought maybe that is caused by air exhaust while talking

No I really have no further idea what to do. The strange thing is that I have the same issue with another telephone which has the exact same setup. So I built two of these with the same hardware and now have the same issue with both. 

I would really appreciate if you have further tipps for me. I really would like to have a more crystal clear sound. 

Big thanks in advance!
Dominik

jaypozo, 2022-09-28:
This happens to me too sometimes. I don't have a solution but I am glad you checked the mic because that is what I was going to do next.

ssnens, 2023-03-10:
Did you have any luck with this issue? I am having the exact same issue and running out of ideas on what it could be.

henna-5, 2023-03-16:
Bumping this here - any solutions have been found yet?

diabloroker, 2023-03-17:
Can everyone post an example? I’ve ran into this issue as well and actively trying to find a solution.