playfultechnology/audio-guestbook #15: Recorded .wav files skipping

amendochka, 2022-09-26:
Hello!

Thanks so much for developing this awesome code!

We managed to connect everything seemingly correctly. The greeting.wav message works on pickup, gives a beep, and it takes in a recorded message. However, when trying to play back that recorded message on the computer, it skips, so we hear like every 3rd word. Any ideas as to why this might be happening?

Thank you!
Amanda

amendochka, 2022-09-26:
to add, comparing the actual recording length (manually with a timer) v. the length of the .wav file recording is ~2:1.

neecx, 2022-09-26:
@amendochka what type of memory card are you using? What's the size and class? 

Try using a 32gb Class 10 if you aren't already

amendochka, 2022-09-27:
Yes! I'm using a SanDisck Ultra Plus, Class 10, 128gb.

neecx, 2022-09-27:
@amendochka yeah that's the problem, the size can't be any more than 32gb 😊

amendochka, 2022-09-27:
ahh thank you @neecx ! I tried reformatting the sd card to FAT32 and it didnt seem to work. I will try with a 32gb and will update

amendochka, 2022-09-27:
The sandisk ultra 32gb didnt seem to work natively, either. And it didn't seem to work when reformatting to FAT32. :

neecx, 2022-09-27:
Is that sandisk ultra 32gb still a class 10? What's your code?

amendochka, 2022-09-27:
yeah it is still a class 10. Using the same exact code as posted in the DD4WH repo (forked from this one for Teensy 4.1 compatibility): https://github.com/DD4WH/audio-guestbook

One thing we noticed debugging, is in the continueRecording function, we added this code:
```
   elapsedMicros usec = 0;
   frec.write(buffer, 512);
   Serial.print("SD write, us=");
   Serial.println(usec);
```

The usec print out might offer some clues. Most of the usecs ~ 13, however, every so often, usec outputs 1000004, and the skips are ~ 1 second in length. No idea how to fix this though and any insight would be appreciated!

Thank you so much for taking the time to help!!

neecx, 2022-09-27:
Strange. Have you tried opening the serial monitor in The Arduino software whilst the phone is plugged in? I'm super new to all of this so didn't know about it. I was then able to figure out at what point in the code something was going wrong.

Might be worth watching the serial monitor whilst picking up the phone to see if any issues are thrown up there

amendochka, 2022-09-27:
yeah! that's where we been printing out all of the usec info, etc. 

No worries though, thank you so much for your help!!

Also if @DD4WH has any input, would really appreciate it!

amendochka, 2022-09-28:
Okay, figured it out. For those having the same issues:

added this in the preamble:
```
#define SPI_CLOCK SD_SCK_MHZ(50)
#ifndef SDCARD_SS_PIN
const uint8_t SD_CS_PIN = SS;
#else  // SDCARD_SS_PIN
// Assume built-in SD is used.
const uint8_t SD_CS_PIN = SDCARD_SS_PIN;
#endif  // SDCARD_SS_PIN

// Try max SPI clock for an SD. Reduce SPI_CLOCK if errors occur.
#define SPI_CLOCK SD_SCK_MHZ(50)

// Try to select the best SD card configuration.
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)
#endif  // HAS_SDIO_CLASS
```
(it is from the SdFat examples)

and changed `SD.begin(SDCARD_CS_PIN)` to `SD.sdfs.begin(SD_CONFIG)`

samygeede, 2022-12-02:
Thanks @amanda-pagul that helped so much!!
Without it the recordings were skipping so bad and now it works like a charm.

Do you by any chance now if the MTP feature works on Mac also?
The Teensy/SD-card does not show up in the file explorer or in the harddrive-menu...

amanda-pagul, 2022-12-03:
awesome, glad it worked! :)

Also we used the phone for our wedding, and just some pro-tips:
- when setting up the recording at home, we originally thought the message was too loud and lowered the volume. _Keep it loud_. Assuming there is music at your event, it will be very hard to hear otherwise.
- we attached an external battery pack via USB to power it (same ones that can power your phone) instead of physical batteries. I think if the phone is not in use, the battery pack might go to sleep so you have to check it is turned on periodically throughout the event, or troubleshoot/test beforehand to see if it goes to sleep.

samygeede, 2022-12-03:
Again, great tips! Thank you! :-)

I was thinking about possbilities to rent out the phone and give the customer the option to replace tha standard prompt with a custom message recorded directly on the phone.

My idea: Have a standard propmt "greeting.wav" and a a function where the customer can press and hold the button _before_ lifting the handset.
He would then be prompted to
a) record a custom greeting "custom_greeting.wav" after the beep
b) press the button again to delete the custom recording (delete file) or
c) hang up to cancel.
The teensy would simply play the custom_greeting.wav if available and otherwise the greeting.wav

Does the process make sense? Do you think it would be complicated to code as I am just beginning?

amanda-pagul, 2022-12-04:
Awesome :)

The easiest easiest might be to ask the customer in advance whether they
want to record a greeting and send a phone with the relevant configuration.

You could probably create a similar function to “playLastRecording” to do
what you want to do. Or you could use the startRecordingGreeting where
there are two .wav files. You might want to include an
invitation_greeting.wav that prompts the user—e.g. if they want to keep the
greeting you have, just hang up, otherwise, hold down the button (you mean
like a digit/number button right?) for 10 seconds to record your own. I
think that you could include that functionality with the timer. But if the
customer messes up or changes their mind, you’d have to add a reset
function too by reusing some of those functions mentioned above and it
becomes trickier (maybe hooking up more buttons to confirm or restart 😅)

Anyway, I think it’s doable but not super trivial. If you add an extra
button, it shouldn’t be too much extra code but if you want to reuse the
hook-switch it becomes pretty tricky. I’m not an expert in Arduino coding
though, so others might have better ideas :)

samygeede, 2022-12-06:
I got it to work! :-)
In case you want to check it out:
https://github.com/samygeede/audio-guestbook