playfultechnology/audio-guestbook #19: Issue With Audio/Recording

brianholle, 2022-10-15:
After setting up the Teensy/Audio adapter and connecting it to the phone, I am running into some issues. I am able to confirm that the pickup/hang-up logic is working correctly via the debugger, which is great. 
Where I start having issues is with the audio playback and the recording. While I can see in the debugger that the greeting prompt is being sent to the speaker, I am unable to hear anything on the phone. Secondly, when recordings are made, which I can successfully see on the SD card and listen to with Audacity, all I hear is static. I can attach an example audio file if that would help.

This is my first time soldering pins, so I am concerned that maybe there was an issue with some of the connections. Is anyone familiar with what issues could cause these symptoms? Any assistance here would be greatly appreciated!

alastaira, 2022-10-17:
Your description would suggest indeed that the code is functioning correctly, but that you have a hardware issue with your connections to the speaker/microphone in the telephone handset.

Besides double-checking your wiring, do you have any other small speaker/microphones you could try?

brianholle, 2022-10-18:
@alastaira I tried to swap out the microphone in the handset that I have connected, but it netted the same result. Is there a diagram somewhere that signifies the importance of each of the ~15 or so pins on both sides of the unit? It looks to me like the connection for the microphone and speaker are both good. It is all of the other pins that I am concerned about. Thanks for your reply!

alastaira, 2022-10-18:
You shouldn't need to worry about the individual function of the other pins - they're what's needed for the I2S and SPI connections (for the digital audio data and the SD card reader, respectively) on the audio shield. But that's the purpose of the shield - assuming your soldering of the headers is good, they just slot onto the correct pins.

brianholle, 2022-10-31:
@N1CK00 @alastaira I was able to resolve the issue by checking all of the solder points. I must have had a bad connection with one of the pins. I am not sure which though so don't have anything else to report. A new issue has appeared though. 

Now, all is working well, but the audio that is recorded is very grainy. It is an old phone, so i am not sure if the issue is phone or software related. Is anyone able to diagnose a possible issue from this sample audio clip?

https://mega.nz/file/sWFCFATC#esECd6b1vFLqy_PYE5qabfV_-Y5yV3nFD3mSdW3xOx8