playfultechnology/audio-guestbook #23: FZZ File Issue

Kiel0211, 2022-11-19:
Hi,

I am unable to open the FFZ file to copy and paste the code into Arduino.  If anyone has managed to download it, please could you put a downloadable file on here. 

Thank you

Kiel

alastaira, 2022-11-21:
Run the Teensyduino installer again, and select to install "SerialFlash"

Kiel0211, 2022-11-21:
Thank you for the reply :)

I have installed "Serial Flash" however still the same issue:

Arduino: 1.8.16 (Windows 10), TD: 1.57, Board: "Teensy 4.0, Serial, 600 MHz, Faster, United Kingdom"

C:\Users\Kiel Wilkinson\Downloads\AudioGuestbook__3_.fzz\AudioGuestbook__3_.fzz.ino:30:25: fatal error: MTP_Teensy.h: No such file or directory

Multiple libraries were found for "SD.h"

compilation terminated.

 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

Error compiling for board Teensy 4.0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

alastaira, 2022-11-21:
That's a different error...  fatal error: **MTP_Teensy.h**: No such file or directory.

Have you installed the MTP_Teensy library?

BillieRubin, 2022-12-20:
I have the same issue.  Where can I get the MTP_Teensy Library.

cdgrammer, 2023-01-05:
> I have the same issue. Where can I get the MTP_Teensy Library.

I was having the same issue. I found this link and made a folder under the hardware\teensy\avr\libraries and named it MTP_Teensy. I added all of the files on this link in that folder and it worked for me.

https://github.com/KurtE/MTP_Teensy/tree/main/src

If you don't know how to make these files. Just open up Notepad, copy the file, and save it as the name that's listed.