playfultechnology/audio-guestbook #24: Error in Arduino IDE after compiling

manni0815held, 2022-11-19:
Hello, 

i have a problem after compiling your code, i use the 4.1 with the Audioshield. Can you help me somehow?

Arduino IDE 2.0.2
MPT_Teensy: https://github.com/KurtE/MTP_Teensy (the same error with the original)

This is what i got:

```
c:\Users\xxxxxxxxx\Privat\Dokumente\Arduino\libraries\MTP_Teensy-main\src\MTP_Teensy.cpp: In member function 'uint32_t MTP_class::GetObjectInfo(MTP_class::MTPContainer&)':
c:\Users\xxxxxxxx\Dokumente\Arduino\libraries\MTP_Teensy-main\src\MTP_Teensy.cpp:523:34: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
   #pragma GCC diagnostic ignored "-Wformat-truncation" /* Or "-Wformat-truncation" */
                                  ^
Opening Teensy Loader...
Memory Usage on Teensy 4.1:
  FLASH: code:110912, data:14636, headers:8592   free for files:7992324
   RAM1: variables:19264, code:107352, padding:23720   free for local variables:373952
   RAM2: variables:45440  free for malloc/new:478848
```

alastaira, 2022-11-21:
I'm not familiar with MTP_Teensy, but what you've posted there is a warning, not an error - it should continue to compile and upload as normal.

criton01, 2023-01-05:
same issues... 

C:\Users\XXXXXX\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:30:24: fatal error: MTP_Teensy.h: No such file or directory
compilation terminated.
Multiple libraries were found for "SD.h"
 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Error compiling for board Teensy 4.0.

When you comment the MTP_Teensy.h out of the included you get this...

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:49:1: error: 'AudioPlaySdWavX' does not name a type
 AudioPlaySdWavX              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file
 ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:54:28: error: 'playWav1' was not declared in this scope
 AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer
                            ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void setup()':
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:140:5: error: 'MTP' was not declared in this scope
     MTP.begin();
     ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void loop()':
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:185:7: error: 'playWav1' was not declared in this scope
       playWav1.play("greeting.wav");    
       ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:238:3: error: 'MTP' was not declared in this scope
   MTP.loop();  // This is mandatory to be placed in the loop code.
   ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void setMTPdeviceChecks(bool)':
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:245:5: error: 'MTP' was not declared in this scope
     MTP.storage()->set_DeltaDeviceCheckTimeMS(MTPcheckInterval);
     ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:250:5: error: 'MTP' was not declared in this scope
     MTP.storage()->set_DeltaDeviceCheckTimeMS((uint32_t) -1);
     ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playAllRecordings()':
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:374:7: error: 'playWav1' was not declared in this scope
       playWav1.play(entry.name());
       ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:380:13: error: 'playWav1' was not declared in this scope
     while (!playWav1.isStopped()) { // this works for playWav
             ^
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playLastRecording()':
C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:411:7: error: 'playWav1' was not declared in this scope
       playWav1.play(filename);
       ^
Multiple libraries were found for "SD.h"
 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Error compiling for board Teensy 4.0.

The original code in the video example makes sense but the new is not so much...  Wave playback is not used?
Help.... need help bigtime on this

michaelhopkins42, 2023-01-09:
Hey @criton01 and @manni0815held 

Found this solution/post for the MTP_Teensy.h issue

https://github.com/playfultechnology/audio-guestbook/issues/13

CC: @alastaira

criton01, 2023-01-31:
have same issue...
Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "Teensy 4.0, Bigger blocks (256 samples), Serial + MTP Disk (Experimental), 600 MHz, Faster, US English"

Warning: Board arduino:avr:teensy40 doesn't define a 'build.board' preference. Auto-set to: AVR_TEENSY40

Warning: Board arduino:avr:teensy41 doesn't define a 'build.board' preference. Auto-set to: AVR_TEENSY41

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:49:1: error: 'AudioPlaySdWavX' does not name a type

 AudioPlaySdWavX              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file

 ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:54:28: error: 'playWav1' was not declared in this scope

 AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer

                            ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void loop()':

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:185:7: error: 'playWav1' was not declared in this scope

       playWav1.play("greeting.wav");    

       ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:238:3: error: 'MTP' was not declared in this scope

   MTP.loop();  // This is mandatory to be placed in the loop code.

   ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void setMTPdeviceChecks(bool)':

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:245:5: error: 'MTP' was not declared in this scope

     MTP.storage()->set_DeltaDeviceCheckTimeMS(MTPcheckInterval);

     ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:250:5: error: 'MTP' was not declared in this scope

     MTP.storage()->set_DeltaDeviceCheckTimeMS((uint32_t) -1);

     ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playAllRecordings()':

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:374:7: error: 'playWav1' was not declared in this scope

       playWav1.play(entry.name());

       ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:380:13: error: 'playWav1' was not declared in this scope

     while (!playWav1.isStopped()) { // this works for playWav

             ^

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playLastRecording()':

C:\Users\crito\Desktop\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:411:7: error: 'playWav1' was not declared in this scope

       playWav1.play(filename);

       ^

Multiple libraries were found for "SD.h"

 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

Error compiling for board Teensy 4.0.

manni0815held, 2023-02-04:
Everything seems to work fine now. It compiled, but... the audiorecordings are choppy.
Like every second there is a second or so of audio cut out.