playfultechnology/audio-guestbook #25: Audio recording problems

zachm520, 2022-11-21:
Hello, I tried to read through everyone else's problems before posting, my audio files are recording blank. I checked my solder connections and they seem ok, but obviously I am a newbie. Im using a teensy 4.1 on the audio shield  (so I do have overhanging pins from the longer 4.1 board). using the serial monitor, it all appears to working, however I do not have anything coming through on the handset. I did try the same setup from the video with 3.5 breakout board, etc. Desperate for help as my wedding is saturday.

alastaira, 2022-11-21:
So the files are being written ok to the SD card, and they've of the correct length for the recording? Are they actually _blank_, or do you simply not hear any audio?

zachm520, 2022-11-21:
So they are .wav files that are recorded that are the correct length, though audacity shows no sound or static being on them, im going to try and solder in another microphone to see if that is the issue

zachm520, 2022-11-21:
I replaced the microphone and that fixed it, where can I add a default greeting to the speaker? thats the one part Im confused on

alastaira, 2022-11-22:
Oh great, glad that fixed it and thanks for letting me know!

To record the greeting, I used audacity on my PC, but you could use any sound recording app on your mobile phone etc., then just copy it to the SD card and rename it to "greeting.wav"

closed 2022-11-22 by alastaira