playfultechnology/audio-guestbook #30: 'AudioPlaySdWavX' does not name a type

fivefortyfour, 2023-01-06:
Why am I getting these compile errors?

`Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "Teensy 4.0, Serial + MTP Disk (Experimental), 600 MHz, Faster, US English"

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:49:1: error: 'AudioPlaySdWavX' does not name a type

 AudioPlaySdWavX              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file

 ^

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:54:28: error: 'playWav1' was not declared in this scope

 AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer

                            ^

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void loop()':

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:185:7: error: 'playWav1' was not declared in this scope

       playWav1.play("greeting.wav");    

       ^

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playAllRecordings()':

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:374:7: error: 'playWav1' was not declared in this scope

       playWav1.play(entry.name());

       ^

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:380:13: error: 'playWav1' was not declared in this scope

     while (!playWav1.isStopped()) { // this works for playWav

             ^

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playLastRecording()':

C:\Users\conce\Downloads\audio-guestbook-main\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:411:7: error: 'playWav1' was not declared in this scope

       playWav1.play(filename);

       ^

Error compiling for board Teensy 4.0.

Invalid library found in C:\Program Files (x86)\Arduino\libraries\SDBackup: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\SDBackup

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

mpetti1816, 2023-01-23:
I just posted on the board about a bunch of problems but I had this same issue:

Inside your sketch, go to the "Sketch" dropdown at the top, choose "Add File" and add play_wav_sd.cpp and repeat with play_wav_sd.h files that were in the Github code download.
Leave those tabs open, then try Verifying your sketch and you should lose those errors.

fivefortyfour, 2023-01-23:
Thanks, yes I’ve done that. 
many idea where you put the greeting audio file? Does it go on the sd card?

mpetti1816, 2023-01-23:
I dropped it onto the SD card in the main directory. After I got the MTP file transfer portion working properly, I see the Teensy (with SD card installed) on my computer as a new Drive when plugged in via USB, with the name I specified in the code. If I open that, the greeting.wav file is there in the top level directory.

aval158, 2023-01-23:
I had the same issue. I solved it by opening play_sd_wav.cpp and play_sd_wav.h as separate tabs in Arduino.