playfultechnology/audio-guestbook #31: Nothing happens after uploading to Teensy

cdgrammer, 2023-01-09:
I was finally able to get all the way through the sketching and uploading. I have a file named Greeting.wav on the sd card, but when I pick up the phone, nothing happens. I have the Teensy 4.0, and the Audio Shield version C. Should the led on the teensy blink while the program is running? Is there a way I can check to see if the teensy and the audio shield are working together correctly? I only have 2 weeks left before we leave for our wedding so I'm hoping I can get this to work before then.

G8Titan, 2023-01-10:
Which version of the code did you use? Did you press the button on the teensy when you finished compiling the code?

cdgrammer, 2023-01-10:
I used the new updated code. I noticed that I have the Audio Board Revision C, and to connect it directly, I need the Revision D. I bought the D and it's on it's way, but I wired it together, and I get a noise when I pick up the phone, but it sounds like what connecting to the internet sounded like in the 90's. It is not playing the Greeting.wav file I have on the sd card.