playfultechnology/audio-guestbook #32: Need updated working code

G8Titan, 2023-01-10:
After watching the video and following along with the code you provided on the website, I've noticed some slight differences between the youtube video and what's provided on the website. Is the code on the website more up-to-date than the one shown in the video? Also, is this a plug-and-play code or will it need tweaking?

mpetti1816, 2023-01-23:
The code on Github has been updated by several users to make improvements, so it doesn't match the video. It will also require some additional tweaks, so not plug-and-play, but also limited changes to the code itself.

I just posted a bunch of problems I ran into and how I fixed them.

aaronwbrown, 2023-06-18:
Hi, @mpetti1816 - where did you post these problems, in the repo issues?