playfultechnology/audio-guestbook #33: Help me with 'AudioPlaySdWavX" error

lilredmom, 2023-01-11:
I have no coding experience, but I'm trying to work through the errors that come up. Now I have this one: 

"Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "Teensy 4.0, Serial + MTP Disk (Experimental), 600 MHz, Faster, US English"

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:49:1: error: 'AudioPlaySdWavX' does not name a type

 AudioPlaySdWavX              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file

 ^

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:54:28: error: 'playWav1' was not declared in this scope

 AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer

                            ^

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void loop()':

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:185:7: error: 'playWav1' was not declared in this scope

       playWav1.play("greeting.wav");    

       ^

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playAllRecordings()':

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:374:7: error: 'playWav1' was not declared in this scope

       playWav1.play(entry.name());

       ^

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:380:13: error: 'playWav1' was not declared in this scope

     while (!playWav1.isStopped()) { // this works for playWav

             ^

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino: In function 'void playLastRecording()':

C:\Users\thebe\AppData\Local\Temp\Temp3_audio-guestbook-main.zip\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:411:7: error: 'playWav1' was not declared in this scope

       playWav1.play(filename);

       ^

Error compiling for board Teensy 4.0."

Anyone can help me?

BrandonJam, 2023-01-22:
Currently have this same issue.  Hope to see a comment, or I will update if I get beyond this.

mpetti1816, 2023-01-23:
I just posted on the board about a bunch of problems but I had this same issue:

Inside your sketch, go to the "Sketch" dropdown at the top, choose "Add File" and add play_wav_sd.cpp and repeat with play_wav_sd.h files that were in the Github code download.
Leave those tabs open, then try Verifying your sketch and you should lose those errors.