playfultechnology/audio-guestbook #34: So close! probably an easy solution - hook switch is reversed

remlazer, 2023-01-13:
after great stress, my phone is nearly all together and working.
my only problem is the phone hook activates the wrong way round - so its recording if I'm hanging the phone up. 
is there an easy bit in the code to switch this around? or is a hardware issue I need to go back in and sort.

its a GPO746 phone is anyone can help! 

great project too! really enjoyed learning a little of code. thanks!

remlazer, 2023-01-13:
was being a nugget infact

closed 2023-01-13 by remlazer

paulhaughey, 2023-02-04:
Hi @remlazer - I have the same issue where the hook activates the opposite was around (I'm being a nugget too clearly) - how did you manage to resolve it?

avierose, 2023-02-11:
@paulhaughey go through the code & when there is a reference to 
```
buttonRecord.fallingEdge()
``` 
swap it for
```
buttonRecord.risingEdge()
``` 
& vice versa. This should fix it for you.

remlazer, 2023-02-11:
Which phone are you using? I figured out on my phones that it doesn’t actually matter once it’s assembled

paulhaughey, 2023-02-11:
Sorry - I should have updated my comment - I had tried changing them but it was actually my bounce rate - I had to update this as there was a lot of “bounce”