playfultechnology/audio-guestbook #36: Several Problems and Solutions I Found During My Build

mpetti1816, 2023-01-23:
It took me a lot of hunting around to fix some of the code issues I was encountering and I'd thought I'd explain my solutions to those of you who are struggling with the same problems. The answers were pulled from this project's main page, the "Closed Issues" board, and a few other places on the internet, but here they are consolidated.

My experiences only pertain to the Github code for this project as of January 22, 2023 and the specific hardware/software that I used.

**My hardware:**
Teensy 4.0 board
Teensy4_AUDIO Audio Shield
Momentary pushbutton
Vintage telephone

**My software:**
Arduino IDE 1.8.19
Teensyduino 1.57 using Windows installer

**First steps:** I did all of the installation adjustmets included on the audio-guestbook Github main page, such as
- copied boards.local.txt
- copied platform.txt
- Changed "Audio tweaks" to "Bigger blocks (256 samples)"

I then had several compiling errors that I had to hunt around for solutions to

**Problem:**
AudioPlaySdWavX does not name a type
&
'playWav1' was not declared in this scope
**Solution:** 
Inside your sketch, go to the "Sketch" dropdown at the top, choose "Add File" and add play_wav_sd.cpp and repeat with play_wav_sd.h files that were in the Github code download.
Leave those tabs open, then try Verifying your sketch and you should lose those errors.

**Problem:**
Multiple libraries were found for "SD.h"
**Solution**:
I cut/pasted the folder 'C:\Program Files (x86)\Arduino\libraries\SD' out of the Arduino libraries directory and saved it elsewhere on my computer for safe keeping. Now, the only 'SD' folder should be under Arduino\hardware\teensy\avr\libraries\

**Problem:**
Compiling error involving SerialFlash.h
**Solution:**
Re-ran the TeensyDuino installer and added the 'SerialFlash' library

**Problem:**
Multiple errors involving MTP_Teensy.h
**Solution:**
Download the Github code from https://github.com/KurtE/MTP_Teensy and unzip the folder into 'C:\Program Files (x86)\Arduino\libraries\'
Then, in your Arduino sketch, select from the dropdowns at the top "Tools > USB Type > Serial + MTP Disk (Experimental)"

**Problem:**
Audio was clipping (breaking up) during playback after I got everything "working"
**Solution:**
There are two volume adjustments in the code that I fiddled with. Use CTRL+F in the code to find "sgt15000_1.volume(0.95)". The 0.95 is 95% volume on the headset speaker, which was too loud. I dropped it down to 0.7 and it stopped clipping on playback.
I also adjusted the "sgt15000_1.micGain(5)", which is how loud the microphone on the headset is recorded in the file. I dropped it to (4) because the recordings sounded blown out.

**Problem:**
My greeting message wasn't playing back, I would just get a single "beep" when I lifted the handset
**Solution:**
I downloaded Audacity and opened my greeting message, then exported it using 44.1kHz and 16-bit PCM as a new WAV file. I also had a stereo recording, so I had to delete one of the tracks to make it "mono". That fixed the playback problem.

**Problem:**
The code compiled correctly (finally), but the handset wasn't responding properly and I was hearing a constant clicking noise in the speaker.
**Solution:**
My phone had a lot of shared terminals in the internal wiring. I disconnected all of the wires from the phone's terminals and directly wired each wire from the speaker/microphone/handset contact/pushbutton to the inputs/outputs of my Teensy board. This solved the clicking issue and after troubleshooting my code, it also solved the audio issues.

fivefortyfour, 2023-01-23:
Nice, thanks for your update! Where do you put the greeting audio file?

BrandonJam, 2023-01-23:
Ok.  So I have board soldered. Code compiled and uploaded to the Board.  I add power to the Board and I hear a beep.  After that, I get nothing.  Not sure what the issue is.  Is there a way to see a log of the steps of code the Board is going through?  Maybe then I can determine where my issue is.

mpetti1816, 2023-01-23:
> Ok. So I have board soldered. Code compiled and uploaded to the Board. I add power to the Board and I hear a beep. After that, I get nothing. Not sure what the issue is. Is there a way to see a log of the steps of code the Board is going through? Maybe then I can determine where my issue is.

That initial beep is an indication that it has power and is running (at least that's what it seems like, mine does the same beep upon power). If you look under the Tools menu, there is something called the Serial Monitor. When you open that and run your board, there are sections of the code that will print out what is going on while it's running. I used that to troubleshoot a few wiring problems I had.

BrandonJam, 2023-01-23:
I just did Serial Monitor.  It keep saying "unable to access SD card".  SD card is definitely present though.

mpetti1816, 2023-01-23:
> I just did Serial Monitor. It keep saying "unable to access SD card". SD card is definitely present though. It is in the same way after programming the board.

I am no where near an expert on this stuff, but a couple things that come to mind. Ensuring your MTP_Teensy libraries were put in the right folders and that "Serial + MTP Disk (Experimental)" was selected (both in my initial post). The other thing to check is that your SD card is not larger than 2GB. That limitation is mentioned at the bottom of this page https://www.pjrc.com/store/teensy3_audio.html

Other than that, I can't think of anything else, so good luck Googling or having someone else chime in.

BrandonJam, 2023-01-23:
I saw another Issue down below that said it was the pin locations with Rev D of the Audio Shield.  I changed those to match the Rev D diagram, but still getting the same message.

cpereirauy, 2023-01-24:
> > I just did Serial Monitor. It keep saying "unable to access SD card". SD card is definitely present though. It is in the same way after programming the board.
> 
> I am no where near an expert on this stuff, but a couple things that come to mind. Ensuring your MTP_Teensy libraries were put in the right folders and that "Serial + MTP Disk (Experimental)" was selected (both in my initial post). The other thing to check is that your SD card is not larger than 2TB. That limitation is mentioned at the bottom of this page https://www.pjrc.com/store/teensy3_audio.html
> 
> Other than that, I can't think of anything else, so good luck Googling or having someone else chime in.

I formatted my SC card (32gb Samsung) to exfat, not saying that would solve your issue, but worth trying.
Another thing, where are you inserting your SD card? Teensy or Audio Shield? Try swithching them

BrandonJam, 2023-01-25:
> When I compile to the teensy I’m getting an error about Serial port not
> selected.

Did you select the port under Tools?

BrandonJam, 2023-01-25:
I am using Teensy 4.0 so SD only in Audio Shield.

I ended up using a different fork version of the code and it worked fine
with my sd card as is. Weird.

fivefortyfour, 2023-01-25:
Here's my error regarding the Serial port issue.... I have tried both the Serial+MPT Disk (experimental) and MTP Disk (experimental)

Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "Teensy 4.0, MTP Disk (Experimental), 600 MHz, Faster, US English, Bigger blocks (256 samples)"

C:\Program Files (x86)\Arduino\libraries\MTP_Teensy\src\MTP_Teensy.cpp: In member function 'uint32_t MTP_class::GetObjectInfo(MTP_class::MTPContainer&)':

C:\Program Files (x86)\Arduino\libraries\MTP_Teensy\src\MTP_Teensy.cpp:523:34: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]

   #pragma GCC diagnostic ignored "-Wformat-truncation" /* Or "-Wformat-truncation" */

                                  ^

Memory Usage on Teensy 4.0:

  FLASH: code:106616, data:13540, headers:8864   free for files:1902596

   RAM1: variables:18912, code:104024, padding:27048   free for local variables:374304

   RAM2: variables:50528  free for malloc/new:473760

Serial port not selected.

Invalid library found in C:\Program Files (x86)\Arduino\libraries\SDBackup: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\SDBackup

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

fivefortyfour, 2023-01-25:
Finally, all working!

rich-day, 2023-01-25:
I had to amend my SD card pins in the code as I am using Audio Shield Rev D - this ended up working (as well as making sure my pins were connected accordingly)
`#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  11
#define SDCARD_SCK_PIN   13`

BrandonJam, 2023-01-25:
Thanks, Rich.  So, I have done this with a few different variations of the code, and also by changing the pin numbers in the code for those different variations, because I too saw that the PIN locations for the T4.0/Audio Shield were different than what people were using.  

https://www.pjrc.com/store/teensy3_audio.html

I ended up getting the DD4WH fork code to work WITHOUT changing the pin locations, which confuses me somewhat as to why that worked if the pin locations are off.

Question back to you though... so I know how to easily change the pin locations in the code, but I wasn't thinking about physically changing the pin connections on the board.  I was under the impression that when the T4.0 and Audio Shield were stacked together, then that is where the pins lined up appropriately based on the link above.  Can you touch on this part a little more?  Maybe even a pic of your board?

rich-day, 2023-01-25:
Sorry, to clarify, I just meant confirming that my pins (10,11,13 in this case) were soldered well and that the board was stacked correctly (e.g. not misaligned or upside down etc). I am by means no expert on this, I've just recently completed this project and ran in to similar problems so just sharing my learnings.

Glad to hear you got it working however!

DevlinDiener, 2023-02-13:
> I am using Teensy 4.0 so SD only in Audio Shield. I ended up using a different fork version of the code and it worked fine with my sd card as is. Weird.
Can I ask which fork you ended up using? I am facing the same problem.

BrandonJam, 2023-02-14:
the fork version from @DD4WH

fivefortyfour, 2023-02-19:
I'm recording my wav files to sd card but when I plug teensy into laptop with usb cable, I can't play the files as it says they're in use? I can't copy them either. How do you use them?  I want to access them with usb cable and not take out the sd card.

TWK-Nattttttttt, 2023-02-21:
I'm a little confused. I downloaded the MTP_Teensy documents that you said to and put them in the libraries folder, but I'm still getting "'MTP' was not defined in this scope." 

The extent of my knowledge with code is drawing a circle on Khan Academy, so I'd appreciate any help I can get *imagine there is a laughing emoji here*

fivefortyfour, 2023-03-05:
I have a phone that has been working fine. Now it seems to have stopped
working. I don't get the 4 beeps when I hang up the phone?
Anyone have this problem?

markmak77, 2023-03-12:
Does anyone know if the author (@alastaira) is still accepting Pull Requests? It'd be nice to add something like this to a "Troubleshooting" section of the readme...