playfultechnology/audio-guestbook #41: Clarify recording wav files saving to SD card

BrandonJam, 2023-01-25:
Have one of the fork versions (DD4WH) of this code working seemingly fine.  

One main question I have related to the saving of the recording wav files to the SD card...
-I have been testing my audio guestbook while connected to my PC for now.
-It seems that the recordings don't save to my SD card right away.  I have the SC card open while operating so I can watch this.
-They do seem to be saving to memory because I can play them back on the phone after recording.
-Once I disconnect the phone from the PC and reconnect it, then the files that were recorded prior appear on the SD card as viewed from the PC.

Is this expected behavior, or should the files be visible as saved on the SD card in relatively real time?

BrandonJam, 2023-01-30:
@DD4WH would you be able to provide some insight for me?  Thanks.

DD4WH, 2023-02-01:
Your test conditions are not valid :-).

* Use the latest software version from my github
* never try to record while the Telephone is connected to your PC !
* use a USB power bank or a power supply to power the telephone
* Record
* unplug power from the telephone
* plug the telephone into your PC
* the recordings should be on your SD card and visible in the file explorer of your PC

YOU CANNOT RECORD WHILE THE TELEPHONE IS PLUGGED INTO THE PC

BrandonJam, 2023-02-01:
@DD4WH Thank you.

I am using your latest version.  Also, do you mean to say "you should not record while the telephone is plugged into the PC", rather than "you cannot"?  I read them differently, and I am pretty sure you can record when it is connected to the PC (it will record), but maybe I/we shouldn't be as it causes potential issues.  

While I have you, I made another issue post where I was trying to clarify the correct external power voltage to use.  In the documentation, I see 3.3v, but I see a bunch of people saying 5v.  Thoughts?

DD4WH, 2023-02-01:
The Teensy documentation is pretty clear where to apply which voltage :-). Read carefully on the PJRC page ;-).

BrandonJam, 2023-02-01:
@DD4WH 

So, based on below... if I am now reading correctly, if you power by USB, you can use 5v, since the board converts 5v to 3.3v when using the VUSB/VIN Pin.  I think I read that too fast the last time around.  Can you confirm I am reading and understanding correctly now?  Would be appreciated.  

**USB Power**
Normally Teensy is powered by your PC or USB hub, through a USB cable. The **_USB power arrives at the VUSB pin, which is connected VIN and powers the entire board._**
**VIN Pin**
When USB power is not used, 5V power may be applied to the VIN pin. Because VIN & VUSB are connected, power should not be applied to VIN while a USB cable is used, to prevent the possibility of power flowing back into your computer. Alternately, a pair of pads on the bottom side may be cut apart, to separate VUSB from VIN, allowing power to be safely applied while USB is in use. (TODO: VUSB-VIN pads photo, right side)
**3.3V Power**
Teensy 4.0 **_has a voltage regulator which reduces the 5V VUSB / VIN power to 3.3V for use by the main processor_** and most other parts. Additional circuitry may be powered from the 3.3V pin. The recommended maximum for external 3.3V usage is 250mA. Teensy 4.0 is not meant to receive power on its 3.3V pin, but this can be done with special modificaton.