playfultechnology/audio-guestbook #43: MTP_Teensy.h

Smithy455, 2023-01-31:
I've read through everything I can find and have downloaded the mtp with the green code button on the site, https://github.com/KurtE/MTP_Teensy I've unzipped it and put in in to programfiles(86): Arduino: libraries then selected serial +MTP Disk. But it comes up with this error? 
not really sure what I've missed? 
[image]

BrandonJam, 2023-02-01:
try putting it in the same folder as your sketch prior to opening it.

Smithy455, 2023-02-01:
Cheers, that didn't seem to work but I've put it in the documents library with the board.txt fill and that seems to have fixed it. now to deal with the new errors after fixing this.