playfultechnology/audio-guestbook #52: Unable to access the SD card

DevlinDiener, 2023-02-13:
After compiling and uploading the code successfully my board doesn't do anything when hooked up to the phone. Naturally I plugged it in and ran the serial monitor to see if I could troubleshoot. The monitor continuously prints the message 

"Unable to access the SD card"

I have the SD card slotted directly onto the audio shield's card slot. I verified the SD card compatibility as well so I'm not sure if this is a code error or a hardware issue. Any help is appreciated!

nrich50, 2023-02-16:
did you ever fix this issue i am having this problem aswell and nothing i have done has fixed it

rhilyard, 2023-02-18:
I think a lot of us are having this same issue. I hope someone can tell us how to fix this. I have tried everything.

BrockV2, 2023-02-18:
Makes 4 of us, I have also run out of ideas. Will keep trying.

nrich50, 2023-02-18:
So after messing with this thing for over 24 hours in my case I had to change the ports since I am using the REV D audio shield and use a different SD card. Idk why my other one wouldn't work but once I put a new one in it was able to access. Bit since I have solved that problem I still have no sound coming from the speaker and no input from the mic

Praxxxer, 2023-02-18:
@nrich50 I'm also using a REV D2 audio shield. What did you change your SDCARD_MOSI_PIN values to?

BrockV2, 2023-02-19:
@Praxxxer 

Please check the PIN numbers you have quoted as they do not match the values on the teensy drawings. 

To help troubleshoot, can you please confirm if you are using Teensy 4.0 or 4.1? 

Please also check the pins for the hook switch as it’s common to see people wire it to ground and 0, not 0 and 1.

DevlinDiener, 2023-02-21:
Update:

I've fixed the "unable to access SD card" issue. For reference, I'm using a Teensy 4.0 and REV D2 audio shield. Here are the pins I've assigned:

- #define SDCARD_CS_PIN    10
- #define SDCARD_MOSI_PIN  7
- #define SDCARD_SCK_PIN   14

I occasionally get this issue during testing (likely due to improper connection), but it will usually fix itself after a reboot.

The problem I have now is my microphone line is not picking up any input, and my speaker doesn't produce output. I assume this is due to a bad soldering job on a few connections. I took a break from this project but am picking it back up. I'll update you all again with any useful info once I've fixed the issue.

rhilyard, 2023-02-21:
Which fork is everyone using? I have the teensy 4.0 and Rev D2 audio shield. I used 
    #define SDCARD_CS_PIN 10
    #define SDCARD_MOSI_PIN 7
    #define SDCARD_SCK_PIN 14
but that didn
t help at all.

nrich50, 2023-02-22:
I believe I am using the fork @DD4WH has. Have you tried a different SD card and making sure it's formatted. I think it needs to be exFAT but I'm not 100% sure. For me I changed all those pins and it still was not working until I got a new SD. And this mag be stupid but I found myself doing this. If you don't have the teensy directly mounted to the audio shield. Make sure the GND on the teensy is also connected to the audio shield GND. I am completely new to all of this so many mistakes have been made lol

nrich50, 2023-02-22:
Still having the issue of no sound or mic input. I thought maybe I messed up the board but since many people are having it then it must be either a bad batch of audio shields or a error in programming

rhilyard, 2023-02-22:
I have used several different types of sd cards  The gnd is connected to the audio shield. I'm completely at a lost on what to do!!

nrich50, 2023-02-24:
i used this for my pins and added a extra one that is not in the original program. it may not fix your issue but its worth a shot.

#define SDCARD_CS_PIN    10
#define SDCARD_MSIO_PIN    12
#define SDCARD_MOSI_PIN  11
#define SDCARD_SCK_PIN   13

rhilyard, 2023-02-24:
@nrich50 Thanks just gave it a try and it didn't help.  Thanks for the suggestion

kyuubii, 2023-02-25:
I also face the same issue "Unable to access the SD card". 
I use a "Teensy 4.0" with an "Audio Shield Rev D2". 
I tried to: 
- use different SD cards (from 512 MB to 64 GB)
- use FAT, FAT32, exFAT, NTFS formats
- use different pins (10, 11, 12, 13)

markmak77, 2023-03-16:
> Update:
> 
> TLDR: Verify the integrity of your hardware connections. Use the pins defined below if you're using Teensy 4.0 + REV D2 audio shield.
> 
> I've fixed the "unable to access SD card" issue. For reference, I'm using a Teensy 4.0 and REV D2 audio shield. Here are the pins I've assigned:
> 
> * #define SDCARD_CS_PIN    10
> * #define SDCARD_MOSI_PIN  7
> * #define SDCARD_SCK_PIN   14
> 
> I occasionally get this issue during testing (likely due to improper connection), but it will usually fix itself after a reboot.
> 
> My problem now is that my microphone line is not picking up any input, and my speaker doesn't produce output. I assume this is due to a bad soldering job on a few connections. I took a break from this project but am picking it back up. I'll update you all again with any useful info once I've fixed the issue.

@DevlinDiener , Thank you for confirming that these are the correct pin associations for Teensy 4.0 + the Rev D2 audio shield. Could you also please share what you have your micro SD card formatted to?

markmak77, 2023-03-17:
**Update: IT WORKS!**
First of all, I'd like to confirm that the pins listed above by DevlinDiener are indeed the correct ones if you're using a Teensy 4.0 + Rev D2 audio shield - my micro SD card initialized successfully and was recognized by the controller using those exact pins. I'm using a class 10 Sandisk MicroSD card, 32 GB, formatted as FAT32 (did a full format, not quick, just in case, although that shouldn't really affect anything in theory).

Secondly, I'd like to suggest to everyone who's had this problem (or will have it in the future): **CHECK YOUR CONNECTIONS/SOLDER JOINTS**! A post [here](https://forum.pjrc.com/threads/70679-Teensy-4-0-Audio-Board-Rev-D-SD-card-problem) got me thinking this might be the issue, and it seems it was. 

I'd soldered headers onto both the Teensy (male) and the audio shield (female), _except_ for the 3 pins on the left side (GND, 0, 1) to leave them available for the button(s). Now, I know the audio shield needs a GND, but I thought it would be ok with the one on the other side, between Vin (aka 5V) and 3.3V (aka 3V), but apparently not... And, to make sure I didn't have any cold solder joints, I took my soldiering iron to each pin (on both the Teensy and the shield) once again, and it seems those 2 actions solved this issue for me. 

PS: before doing any additional soldering, don't forget to first remove the SD card from the shield.

(cc @nrich50 , @rhilyard , @Praxxxer , @kyuubii )

markabruce, 2023-03-17:
Hi everyone,

I am not yet 100% sure why, but for me the `#ifdef TEENSY_41` seemed to be the culprit...

I have a Teensy 4.0 and a Audio Shield Rev D2.

Remove the two references of `#ifdef TEENSY_41`.

**So when defining pins...**

```
#ifdef TEENSY_41
#define SDCARD_CS_PIN    BUILTIN_SDCARD
#else
#define SDCARD_CS_PIN    10
#endif
```

becomes:
```
// Define pins used by Teensy Audio Shield
#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  7
#define SDCARD_SCK_PIN   14
```

**When setting MOSI and SCK pins...**

```
#ifndef TEENSY_41   // assumes that you are using the SD card slot of the AUDIO BOARD
  SPI.setMOSI(SDCARD_MOSI_PIN);
  SPI.setSCK(SDCARD_SCK_PIN);
#endif  
```

becomes:

```
  SPI.setMOSI(SDCARD_MOSI_PIN);
  SPI.setSCK(SDCARD_SCK_PIN);
```

markmak77, 2023-03-17:
@markabruce , that's probably because you had Teensy 4.1 selected in the Arduino IDE under Tools -> Board: "Teensy x.x". To change it (i.e. define TEENSY_40 instead of TEENSY_41), select Tools -> Board: "Teensy x.x" -> Teensyduino -> Teensy 4.0.

markabruce, 2023-03-18:
I am pretty sure I had 4.0 selected, I'll double check tomorrow.

markabruce, 2023-03-18:
Here:

[image]

I have Board: Teensy 4.0 and yet the `IFNDEF TEENSY_41` must somehow skip settings the pins, maybe a bug with my version of Arduino IDE (1.8.19), or the Teensy installer...

I just copied what `File > Examples > Audio > HardwareTesting > SdCardTest` was doing, which is what PRJC recommend: https://www.pjrc.com/store/teensy3_audio.html

The IFNDEF is saying if TEENSY_41 is NOT defined, then set the SPI - so I've not idea how the software must think TEENSY_41 is defined, is it some sort of global variable set by one of the libraries? Can't find any docs on it...

BrockV2, 2023-03-18:
Has to be defined in the sketchbook code not the menu.

[image]<https://user-images.githubusercontent.com/10707880/226093869-4a1a1430-52c8-4878-936c-92de82535e80.png>

markabruce, 2023-03-18:
> Has to be defined in the sketchbook code not the menu. 

Aha! Thank you, I was being a fool and I hadn't removed #define TEENSY_41, I thought I had searched for every reference to it, but obviously didn't! The forked code even mentions to do this in the [README.md](https://github.com/DD4WH/audio-guestbook/blob/main/README.md).

I suspect though, that this could be the issue a lot of other people are facing. Search for every line of `TEENSY_41` and comment it out or remove it, it is not required if you're on a TEENSY 4.0.