playfultechnology/audio-guestbook #54: Stuck in "ready" mode

andrerossi93, 2023-02-15:
Hi all,
I'm using @DD4WH fork here, struggled a bit for setting up my Teensy 4.0 w/ Rev D2 Audio Shield, found #36 and managed to fix some issues, but when I load code to the Teensy I hear the initial "beep" and on Serial Monitor I get those lines

```
Serial set up correctly
Mode switched to:  Initialising
SD card correctly initialized
Add **SDClass** file system
Added SD card via MTP
Mode switched to:  Ready
&&&&& Dump MTPStorage Loop Data &&&&&
	Callback Data:		0	0x0	0
		1	0xd115	1
		2	0x0	0
		3	0x0	0
		4	0x0	0
	File Systems:
		0	0x200035c8	1

```

and nothing happens. Any hints? Thank you in advance!

jhayen219, 2023-03-10:
So even with this I was able to move on to the next "Mode". If you don't have a change in the hook pin, nothing else will happen. What's the hardware? How is the hook pin wired?

jcantsleep, 2023-03-28:
@andrerossi93 any luck? I have the exact same problem. Teensy 4.0 and Audio Shield Rev D. The code compiled fine. When I load the Teensy, I hear the initial beep. I get the exact same lines as you on the Serial Monitor. But I'm stuck in ready. When I press the hook switch, nothing happens. The 2 hook pin wires are wired to the top of the teensy (orange and brown wires pictured). The other end uses spade connectors to connect directly to the hook pin terminals on the rotary phone.

[image]

jcantsleep, 2023-04-03:
@andrerossi93 I figured it out. I had to amend my SD card pins in the code to the below, as I am using Audio Shield Rev D (issue #36). Now when I lift the handle, I hear the initial 'beep', the greeting, and the audio then starts recording.

#define SDCARD_CS_PIN 10
#define SDCARD_MOSI_PIN 11
#define SDCARD_SCK_PIN 13

Hope that works for you.