playfultechnology/audio-guestbook #55: no sound from speaker or input from mic

nrich50, 2023-02-17:
I have the wire from the center of the mic connected to the mic pin and the outside sleve to the gnd pin (i have tried reversing them) but i still dont have any audio on my recordings.

and i do not hear a beep or my greeting once the hook switch is made. serial monitor says that it is prompting then switches to recording but i hear nothing. i used a 3.5mm headphone jack and have tried reversing both the tip and the sleeve wires but still nothing

i know my speaker and mic are working because i can hear a buzz from the speaker when i touch the wires and i put a meter across the mic and when i talk the resistance changes with how loud i speak.

Praxxxer, 2023-02-18:
I am having the same issue.

nrich50, 2023-02-18:
@Praxxxer let me know if you figure it out and I will do the same. I ordered a new audio shield and I got another aux breakout board maybe that's all

Praxxxer, 2023-02-18:
I just ordered a new aux breakout board too. I really want to avoid buying a whole new shield. Are you using those block connectors? If so, what are your speaker connections reading with the multi-meter?

cpereirauy, 2023-02-22:
same here, got a couple of vintage phones and even ordered new mic and speakers, but still the same results.
I was hoping by now I would have this in working order

nrich50, 2023-03-06:
Has anyone found a solution yet?

EricaFitzz, 2023-03-09:
Same issue here, its writing the recordings to the SD but no sound

EricaFitzz, 2023-03-10:
Check that your mic and speaker wires are in the correct places, mine were swapped. Working great now!