playfultechnology/audio-guestbook #59: Noticed something..

z3gineer, 2023-02-23:
Got the code to work. Everything functions perfectly! 

However, I noticed that if you hang-up the phone at the exact time as the "beep" that comes after the greeting message, the program continues to record and does not recognize the phone is placed back on the hook.

Anyone know what can be done to fix this?

I placed a maximum duration limit for recordings in the code to avoid an accidental long recording, but I would still like to solve the problem.

SehlingS, 2023-02-23:
Hi, 

i'm sorry i have no solution for you. 
But can you please share your way to implement a duration limit for the recordings.

Best regards
Sascha

Gaunajr, 2023-02-24:
> Got the code to work. Everything functions perfectly!
>
> However, I noticed that if you hang-up the phone at the exact time as the "beep" that comes after the greeting message, the program continues [...]

What line did you put a limit recording time at? I noticed that as well and it happened where a message was left for an hour due to this same problem. 

Thanks in advance.

z3gineer, 2023-02-24:
So there's 3 sections you need to add code (see pictures below)

the "120000" is the limit in milliseconds (= 2 min)

Also, my handset hook switch was the "reversed version", so my code will look slightly different than the original code posted.

let me know if this helps. 

[image]

[image]

[image]

Gaunajr, 2023-02-28:
> So there's 3 sections you need to add code (see pictures below)
>
> the "120000" is the limit in milliseconds (= 2 min)
>
> Also, my handset hook switch was the "reversed version", so my code will look [...]

Thank you for the reply.
And to clarify, 120000 should be entered in lines 84, 243, and 255. Or just on 255 like shown on the screenshots?

Gaunajr, 2023-02-28:
I added the timer to the code but it is not functioning. Recording time goes over the 2 min mark. By looking at the code below do you see anything that went wrong? 

Thank you in advance.

[code]
/**
 * Audio Guestbook, Copyright (c) 2022 Playful Technology
 * 
 * Tested using a Teensy 4.0 with Teensy Audio Shield, although should work 
 * with minor modifications on other similar hardware
 * 
 * When handset is lifted, a pre-recorded greeting message is played, followed by a tone.
 * Then, recording starts, and continues until the handset is replaced.
 * Playback button allows all messages currently saved on SD card through earpiece 
 * 
 * Files are saved on SD card as 44.1kHz, 16-bit, mono signed integer RAW audio format 
 * --> changed this to WAV recording, DD4WH 2022_07_31
 * --> added MTP support, which enables copying WAV files from the SD card via the USB connection, DD4WH 2022_08_01
 * 
 * 
 * Frank DD4WH, August 1st 2022 
 * for a DBP 611 telephone (closed contact when handheld is lifted) & with recording to WAV file
 * contact for switch button 0 is closed when handheld is lifted
 * 
 * GNU GPL v3.0 license
 * 
 */

#include <Bounce.h>
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <TimeLib.h>
#include <MTP_Teensy.h>
#include "play_sd_wav.h" // local copy with fixes

// DEFINES
// Define pins used by Teensy Audio Shield
#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  7
#define SDCARD_SCK_PIN   14
// And those used for inputs
#define HOOK_PIN 0
#define PLAYBACK_BUTTON_PIN 1

#define noINSTRUMENT_SD_WRITE

// GLOBALS
// Audio initialisation code can be generated using the GUI interface at https://www.pjrc.com/teensy/gui/
// Inputs
AudioSynthWaveform          waveform1; // To create the "beep" sfx
AudioInputI2S               i2s2; // I2S input from microphone on audio shield
AudioPlaySdWav              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file
AudioRecordQueue            queue1; // Creating an audio buffer in memory before saving to SD
AudioMixer4                 mixer; // Allows merging several inputs to same output
AudioOutputI2S              i2s1; // I2S interface to Speaker/Line Out on Audio shield
AudioConnection patchCord1(waveform1, 0, mixer, 0); // wave to mixer 
AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer
AudioConnection patchCord4(mixer, 0, i2s1, 0); // mixer output to speaker (L)
AudioConnection patchCord6(mixer, 0, i2s1, 1); // mixer output to speaker (R)
AudioConnection patchCord5(i2s2, 0, queue1, 0); // mic input to queue (L)
AudioControlSGTL5000      sgtl5000_1;

// Filename to save audio recording on SD card
char filename[15];
// The file object itself
File frec;

// Use long 40ms debounce time on both switches
[... 469 lines cut ...]
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
}

void print_mode(void) { // only for debugging
  Serial.print("Mode switched to: ");
  // Initialising, Ready, Prompting, Recording, Playing
  if(mode == Mode::Ready)           Serial.println(" Ready");
  else if(mode == Mode::Prompting)  Serial.println(" Prompting");
  else if(mode == Mode::Recording)  Serial.println(" Recording");
  else if(mode == Mode::Playing)    Serial.println(" Playing");
  else if(mode == Mode::Initialising)  Serial.println(" Initialising");
  else Serial.println(" Undefined");
}
[/code]

z3gineer, 2023-03-01:
Hey. yea I see the problem.

 It looks like your original code was written with the falling edge occurring when the handset is lifted. Meaning your telephone you used worked similar to the one in the video by the author.

My phone worked in reverse, so I had to implement the "HANDHELD_CLOSES_ON_LIFT" changes.

Send me your complete original code before you made the changes for the "Timer" (the code that worked with your phone) and I will tell you where to insert the Timer code.

Gaunajr, 2023-03-01:
> Hey. yea I see the problem.
>
> It looks like your original code was written with the falling edge occurring when the handset is lifted. Meaning your telephone you used worked similar to the one in the [...]

Thank you for you taking the time to look at it. I spent a couple of hours looking thru it and couldnt figure it out. It uploads fine and works as before but the timer is not kicking in. 

You can find the full file on the link below and you can also upload to it as well to see what the difference was. 

https://u.pcloud.link/publink/show?code=kZTQfPVZbOPdoj85c2fjCTwVzMt8C8JlBCpV

Thank you a million.

z3gineer, 2023-03-01:
Try this and let me know if it works..

/**
 * Audio Guestbook, Copyright (c) 2022 Playful Technology
 * 
 * Tested using a Teensy 4.0 with Teensy Audio Shield, although should work 
 * with minor modifications on other similar hardware
 * 
 * When handset is lifted, a pre-recorded greeting message is played, followed by a tone.
 * Then, recording starts, and continues until the handset is replaced.
 * Playback button allows all messages currently saved on SD card through earpiece 
 * 
 * Files are saved on SD card as 44.1kHz, 16-bit, mono signed integer RAW audio format 
 * --> changed this to WAV recording, DD4WH 2022_07_31
 * --> added MTP support, which enables copying WAV files from the SD card via the USB connection, DD4WH 2022_08_01
 * 
 * 
 * Frank DD4WH, August 1st 2022 
 * for a DBP 611 telephone (closed contact when handheld is lifted) & with recording to WAV file
 * contact for switch button 0 is closed when handheld is lifted
 * 
 * GNU GPL v3.0 license
 * 
 */

#include <Bounce.h>
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <TimeLib.h>
#include <MTP_Teensy.h>
#include "play_sd_wav.h" // local copy with fixes

// DEFINES
// Define pins used by Teensy Audio Shield
#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  7
#define SDCARD_SCK_PIN   14
// And those used for inputs
#define HOOK_PIN 0
#define PLAYBACK_BUTTON_PIN 1

#define noINSTRUMENT_SD_WRITE

// GLOBALS
// Audio initialisation code can be generated using the GUI interface at https://www.pjrc.com/teensy/gui/
// Inputs
AudioSynthWaveform          waveform1; // To create the "beep" sfx
AudioInputI2S               i2s2; // I2S input from microphone on audio shield
AudioPlaySdWav              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file
AudioRecordQueue            queue1; // Creating an audio buffer in memory before saving to SD
AudioMixer4                 mixer; // Allows merging several inputs to same output
AudioOutputI2S              i2s1; // I2S interface to Speaker/Line Out on Audio shield
AudioConnection patchCord1(waveform1, 0, mixer, 0); // wave to mixer 
AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer
AudioConnection patchCord4(mixer, 0, i2s1, 0); // mixer output to speaker (L)
AudioConnection patchCord6(mixer, 0, i2s1, 1); // mixer output to speaker (R)
AudioConnection patchCord5(i2s2, 0, queue1, 0); // mic input to queue (L)
AudioControlSGTL5000      sgtl5000_1;

// Filename to save audio recording on SD card
char filename[15];
// The file object itself
File frec;

// Use long 40ms debounce time on both switches
Bounce buttonRecord = Bounce(HOOK_PIN, 40);
Bounce buttonPlay = Bounce(PLAYBACK_BUTTON_PIN, 40);

[... 455 lines cut ...]
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
}

void print_mode(void) { // only for debugging
  Serial.print("Mode switched to: ");
  // Initialising, Ready, Prompting, Recording, Playing
  if(mode == Mode::Ready)           Serial.println(" Ready");
  else if(mode == Mode::Prompting)  Serial.println(" Prompting");
  else if(mode == Mode::Recording)  Serial.println(" Recording");
  else if(mode == Mode::Playing)    Serial.println(" Playing");
  else if(mode == Mode::Initialising)  Serial.println(" Initialising");
  else Serial.println(" Undefined");
}

z3gineer, 2023-03-01:
I uploaded a new file to your link. 

Try it out and let me know if it works.

Gaunajr, 2023-03-01:
> I uploaded a new file to your link.
> 
> Try it out and let me know if it works.

You are amazing, I dont know what I missed or how you got it to work but it worked. 

I owe you coffee, send me your email address. 

Thanks a million.

z3gineer, 2023-03-01:
hahahah its all good. Glad I could help

sakil1, 2023-03-02:
@z3gineer I can't get my audio guest book to work. Would you be interested in helping me out? I will compensate you for your time

SehlingS, 2023-03-14:
@z3gineer 

Hey agian, i found the main problem of that issue.
During all wait(xxx) commands the buttons (falling.edge and rising.edge) will not be recoginize.
The is a separate function void wait in the code but this will only create a serial console output and no action inside the code.

I have now changed the code in a few places. Now also a hang up during all wait times is recognized.
I will create a new fork today or tomorrow with all my changes.
But there are some other improvements included. It is based on the fork of [DD4WH](https://github.com/DD4WH)

best regards
Sascha

Jeffff34, 2023-07-25:
> I uploaded a new file to your link.
> 
> Try it out and let me know if it works.

Hello,
It's possible to have the file, my phone work in reverse too.

Thank you.

Have a good day.

Best regards