playfultechnology/audio-guestbook #60: Arduino IDE 2.0 Configurations

smaybs, 2023-02-23:
Just getting familiar with the Arduino 2.0 release.  It no longer uses the platform and boards text files and, as such, not clear how to incorporate the recommended configuration adds referenced in the documentation for the audio-guestbook.

nrich50, 2023-02-24:
i am using arduio ide 1.8.19 and have no issue getting the boards to work im only having the issue of no sound or mic input

smaybs, 2023-02-26:
As for the speaker and mic working ... I assume they were functioning pieces to begin with (could be tested) ... you might re-check your leads to make sure they are secure and to the correct wires (I've found that not every manufacturer follows a proper wire color use ... for example I've seen red wires get connected to green, etc.)

RavioliStorm, 2023-02-28:
@smaybs Have you been able to figure this one out? I'm having a similar issue (noob here)

smaybs, 2023-03-01:
Ravioli ... I'm still working my way through things. You might want to consider using the legacy version as the instructions all support it. I am interested in comparing the two to learn.

Jipsey-A, 2023-05-20:
@smaybs did using the legacy version of Arduino work for you?

rickstaa, 2023-09-21:
For me, the config file location in Arduino IDE 2.0 was `C:\Users\<USER>\AppData\Local\Arduino15\packages\teensy\hardware\avr\`. You can find your location by following [this answer](https://forum.arduino.cc/t/arduino-ide-2-0-and-gcc-optimization/920747/2 from the Arduino team.