playfultechnology/audio-guestbook #61: jerky Greeting message ???

LaFabrik02, 2023-02-25:
Hello, I am new with arduino project. Thanks so much for this tuto.
I have issue with the greeting message.
I can hear it but it's jerky. When i read the messages there are no problem.
I tried to record a new greeting message with the phone, and change its name by greeting.wav and it works but i prefer record with my smartphone and put it in the SD card by USB because recorded messages are less clear.

How can I fix it?

cadbury, 2023-03-01:
I have the same problem. Comparing the file recorded on the phone and a file exported from Audacity appear to be identical, settings wise.

Yet the exported file breaks up, and the file from the phone plays perfectly.

cadbury, 2023-03-16:
I accidentally figured out why this is happening today. I was building a 2nd phone, and to my surprise the issue was gone.

Narrowed it down to the handset. The new handset worked fine, the old one had the bad audio. So on a hunch, I went into Audacity and lowered the gain by 9dB and the audio plays fine now.