playfultechnology/audio-guestbook #62: listen 5 or 10 records?

LaFabrik02, 2023-02-25:
Hello. My audio guestbook works fine but i would like to lister 5 or 10 last messages. Is it possible? Listen the last first.
Thanks !

Campos63, 2023-07-07:
I am having this same issue! When I press the button all I hear is the most recent recorded message. I want to be able to listen to tall the past recorded messages. If anyone has a solution to this I would greatly appreciate it.

Handerson91, 2024-03-06:
While I have some experience in coding syntax for statistical software, this is my first actual coding project.  I'm making this for my own wedding, so I was keen to try and add this option to make the phone a more interactive and fun to use on the day.  I haven't completely finished the project yet but I just got this feature working!  

I didn't want to mess with the code too much, so I just modified the current 'play last recording' sequence.  Like in the tutorial, I've installed a play-back button.  In the current code, pressing the button while the greeting.wav is playing will interrupt the 'prompting' sequence (playing the greeting) and start the 'playLastRecording' sequence.  I took this same approach so the user can interrupt the latest message as it's playing by pressing the button again to skip to the previous recording.  

I'm sure there would be a more automated way to code this so you could skip back all the way to the first recording, but I have no idea how to do that.  Instead, I've added the option to go back through the 5 most recent recordings.  Though, once you know the process, you could add it as many times as you like if, for example, you want to be able to hear the 10 most recent messages.  I figured 5 gives me a few I could listen to a few at a time on the day but not ruin the surprise of hearing them all.  

Here at the steps to add this function:

First, find the section of the code that begins with: void playLastRecording() {

Underneath that, you will see the line: uint16_t idx = 0; 
Step 1: Here, you will need to replicate this line to define one new variable for each additional call you want to playback.  I'm playing back 5 calls, so I need to add 5 variables here, so instead of the one line above I now have: 
uint16_t idx = 0; 
uint16_t idx2 = 0; 
uint16_t idx3 = 0; 
uint16_t idx4 = 0; 
uint16_t idx5 = 0; 

Step 2: The next few lines identify the next available file number for recordings (defined as i in the code). Once i is found, the code redefines our above variable of idx to i - 1 (the most recent recording). Here, you need to redefine the additional variables we have added above, so my code now looks like:
idx = i - 1; 
idx2 = i - 2;
idx3 = i - 3;
idx4 = i - 4;
idx5 = i - 5;

At this point, I'd recommend verifying the code to make sure it still works.  You should see a white warning that you have set the four new variables but not used them, but the code should still successfully verify.  If not, you've made an error. 

Step 3: Next you want to copy the following lines of code:
 // now play file with index idx == last recorded file
      snprintf(filename, 11, " %05d.wav", idx);
      Serial.println(filename);
      playWav1.play(filename);
      mode = Mode::Playing; print_mode();
      while (!playWav1.isStopped()) { // this works for playWav
      buttonPlay.update();
      buttonRecord.update();
      // Button is pressed again
//      if(buttonPlay.risingEdge() || buttonRecord.risingEdge()) { // FIX
      if(buttonPlay.fallingEdge() || buttonRecord.risingEdge()) {
        playWav1.stop();
        mode = Mode::Ready; print_mode();
        return;
      }   
    }

Step 4: Then you want to replace (paste over) the following two lines of code, which are within the section you just copied:
mode = Mode::Ready; print_mode();
return;

Step 5: Within the duplicated section you have just pasted, you want to adjust the first two lines to: 
// now play file with index idx2 == 2nd last recorded file
snprintf(filename, 11, " %05d.wav", idx2);

This now tells the code that when the playback button is pressed a second time, to stop the current message playing and play the second most recent recording.  
Now just repeat steps 3 to 5 for as many variables as you created in step 1.  
Once you have finished, you should now have 
- multiple sections of the same code ending with the line: playWav1.stop();
- the two lines of code returning the mode to ready 
- two lines closing the parentheses FOR EACH of your playback variables
- the three lines closing the 'playLastRecording' sequence
I have 5 playback variables so mine looks like:
          mode = Mode::Ready; print_mode();
          return;
          }
          }
        }
        }
      }
      }
    }
    }
  }
  }
  // file has been played
  mode = Mode::Ready; print_mode();  
  end_Beep();
}

Verify this to check there are no errors and you're good to upload and test the new function.

I hope this helps!