playfultechnology/audio-guestbook #63: "Compilation error: MTP_Teensy.h: No such file or directory"

techguy12321, 2023-03-02:
I'm getting an error
"exit status 1"
"Compilation error: MTP_Teensy.h: No such file or directory"

What should I do?

Fanie123456, 2023-03-02:
same here, have you found out? I tried to download the lib and add it to the teensy lib directory. But then i do get a lot of multiple defintion issues..

Fanie123456, 2023-03-02:
Download the MTP teensy lib from here: https://github.com/KurtE/MTP_Teensy 
Unzip it and pack it to your teensy library directory (for me it is C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries)

Do not forget to add play_sd_wav.cpp and play_sd_wav.h to your guestbook arduino project. Not by copying it manually into the folder, but by using arduinos IDE sketch->addFile. 

Then it should compile.

markmak77, 2023-03-12:
@techguy12321, @Fanie123456 - it seems just copying the library is not enough, as you will get the error
`#error "You need to select USB Type: 'MTP Disk (Experimental)' or 'Serial + MTP Disk (Experimental)'"`
However, I found a solution [here](https://github.com/playfultechnology/audio-guestbook/issues/43#issue-1564935021) that at least gets the project to compile [go to/select: Tools -> USB Type -> Serial + MTP Disk (Experimental)].