playfultechnology/audio-guestbook #64: Audio Tweaks not showing up

Olliepoll, 2023-03-08:
Hi 

I've got everything working. I have issues with the sound clipping and breaking up on playback of the recordings. (all fine with the greeting.wav) I was hoping the Audio tweaks setting would fix this but cant for the life of me get it to show up in the tools menu . Im using a Teensyduino on a mac and have copied both .txt files in the correct place i think! Screenshots attached. Can anyone assist ? 

Thanks in advance. 

[image]
[image]

jhayen219, 2023-03-10:
How do the recordings sound when the audio jack is connected to a computer? If it is still choppy, it might be your handset connections. That's a good place to start.

timtelen91, 2023-03-16:
Same issue here!
Just found this after I created my issue!

Olliepoll, 2023-03-17:
> 

Are you using a Mac too ?

tim-telen, 2023-03-20:
> > 
> 
> Are you using a Mac too ?

No, I have Windows

EWeezyPeezy, 2023-10-10:
Had the same issue.  This was the fix:

Upon restart, a [known bug in Arduino IDE 2.0](https://github.com/arduino/arduino-ide/issues/1030) may prevent it from recognizing changes in the boards.txt and boards.local.txt files. If you encounter this issue, delete the C:\Users\<user name>\AppData\Roaming\arduino-ide\ folder for the "Audio Tweaks" to appear correctly.

This is of course for Windows, but I would imagine it would be the same for Mac, just different file location.

0xernesto, 2023-10-16:
I have the same issue, even after deleting `~/Library/Application Support/arduino-ide/`, Audio Tweaks does not show up.