playfultechnology/audio-guestbook #67: Greeting playback on phone noisy/robotic

critchio, 2023-03-12:
I have resolved all issues and everything seems to be working correctly except my greeting message is noisy/robotic. I recorded the WAV file in audacity using mono, 44100 hz, and 16-bit. I have tried adjusting volume down but it did not solve the issue. Thanks for any advice.

cadbury, 2023-03-16:
See my comment here: https://github.com/playfultechnology/audio-guestbook/issues/61#issuecomment-1472672451

Wucainclan, 2023-06-20:
Hey, experiencing a similar issue here with both the greeting, as well as the recordings coming back sounding robotic and either slow or fast. I'm using the original [code](https://github.com/playfultechnology/audio-guestbook/commit/6ea6f970a401c6c35241ee8f399e81da47a57c58) I referenced previous comments and tried those solutions and still can't seem to figure it out. I have also checked my connections by plugging the handset into my laptop and playing music through the speaker, etc.. I recorded my greeting message via audacity and exported at 44100hz, 16bit. I'm also using a teensy 4.0 and audio shield rev d2 with the sd card plugged into the audio shield. 
Any help would be greatly appreciated!

cadbury, 2023-06-20:
In Audacity, choose effects, then amplify. Do -9db and see if it gets any better.

cadbury, 2023-06-21:
After lowering the volume, did it get any better? Or did it stay exactly the same?

Wucainclan, 2023-06-21:
> After lowering the volume, did it get any better? Or did it stay exactly the same?

It stayed exactly the same.

Wucainclan, 2023-06-26:
@cadbury Any other suggestions here? I've posted a new issue in case anyone else has any ideas, but no hits yet. Not sure where to go from here.

gever4ever, 2023-07-03:
Try switching to Normal blocksize (128), it solved the robotic sound to me.
I've yet to find the reason, might want to try the PJRC forums, they have a thread about the project,
https://forum.pjrc.com/threads/70553-Teensy-4-0-based-Audio-Guestbook

Wucainclan, 2023-07-03:
> Try switching to Normal blocksize (128), it solved the robotic sound to me.
> I've yet to find the reason, might want to try the PJRC forums, they have a thread about the project,
> https://forum.pjrc.com/threads/70553-Teensy-4-0-based-Audio-Guestbook

Did you just change that in the tools menu or did you go and adjust it in the code, as well?

gever4ever, 2023-07-04:
> Did you just change that in the tools menu or did you go and adjust it in the code, as well?

Just in the tools menu.

Wucainclan, 2023-07-04:
> Just in the tools menu.

I tried that and am still experiencing it with the recordings. I managed to get the greeting to play fine but recordings are giving me trouble still.

gever4ever, 2023-07-04:
> I tried that and am still experiencing it with the recordings. I managed to get the greeting to play fine but recordings are giving me trouble still.

Could be a faulty / reversed mic wiring?
Also might want to try to play with the micGain. How do they sound on an external player?

Wucainclan, 2023-07-04:
> > I tried that and am still experiencing it with the recordings. I managed to get the greeting to play fine but recordings are giving me trouble still.
> 
> Could be a faulty / reversed mic wiring?
> Also might want to try to play with the micGain. How do they sound on an external player?

I don't think it's reversed mic wiring. I checked the connections and swapped out the wires. I also played with the gain. I don't have a button for playback on the device so the recordings are always being played when I plug the SD card into my laptop. The only thing I can think of is it's an SD card thing since I'm using a Rev D2 audio shield (I changed the pin numbers on the code) or a faulty handset.

themilkman555, 2024-06-12:
I have a similar issue with the recordings being static/robotic while the greeting message is fine. Did you ever find a solution?

cadbury, 2024-06-12:
Adjusting the mic gain completely fixed it for me.

Depending on the phone, I needed different settings to fix the problem.

themilkman555, 2024-06-13:
> Adjusting the mic gain completely fixed it for me.
> 
> Depending on the phone, I needed different settings to fix the problem.

Sorry, I'm pretty new to all this. Mic gain is something you change in the coding, right? Should I increase or decrease it and by how much do you think?

cadbury, 2024-06-13:
In the code. You probably need to lower it.

themilkman555, 2024-06-13:
You're referring to these lines here, right? I tried lowering from 15 to 5 and it didn't seem to make much of a difference?
[image]
It is a pretty old rotary phone, ~50s ish. Could the issue be with the mic there? 
[image]

themilkman555, 2024-06-13:
Also just realized that in the hardware in the updated readme files it lists shielded microphone cable, which I did not use. Could that be affecting my sound quality? I just used the regular wiring I had on hand. 

There is also a AOM5024 electret low noise microphone capsule listed on that hardware list. Could a better microphone possibly solve my issue?

cadbury, 2024-06-13:
I was able to get all mics work. Brand new phones, some from the 90s, 60s.

Try a lower gain 1, or even 0.5

themilkman555, 2024-06-13:
I’ve done trial and error with a range of gains, even setting it very low (gain=0) didn’t fix the static. I’m using a Western Electric model 500 rotary phone.

cadbury, 2024-06-13:
If you remove the SD card and play the file on a computer, does the audio still have problems?