playfultechnology/audio-guestbook #68: Audio tweaks not showing up

timtelen91, 2023-03-16:
After following the procedure with the boards.local.txt and the platform.txt, the audio tweaks are not showing up in my tools. Even after reboot of the Arduino IDE:

I have Arduino IDE 1.8.19.

Everything is working except the quality of the created wav files is bad.

owelchez, 2023-05-18:
Hello @timtelen91,

I have the same problem as you have described, did you find a solution?

EWeezyPeezy, 2023-10-10:
Had the same issue.  This was the fix:

Upon restart, a [known bug in Arduino IDE 2.0](https://github.com/arduino/arduino-ide/issues/1030) may prevent it from recognizing changes in the boards.txt and boards.local.txt files. If you encounter this issue, delete the C:\Users\<user name>\AppData\Roaming\arduino-ide\ folder for the "Audio Tweaks" to appear correctly.