playfultechnology/audio-guestbook #71: Fully Compiled, but audio (greeting, beep) and mic are not working. Teensy 4.1, Audio Shield Rev D

ranasats, 2023-03-23:
Using the latest sketch from here. I'm on a Teensy 4.1 with Audio Shield Rev D. No Major errors when compiling, just this warning: 

`C:\Users\xxxxxx\Documents\Arduino\libraries\MTP_Teensy\src\MTP_Teensy.cpp: In member function 'uint32_t MTP_class::GetObjectInfo(MTP_class::MTPContainer&)':
C:\Users\xxxxxx\Documents\Arduino\libraries\MTP_Teensy\src\MTP_Teensy.cpp:523:34: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
   #pragma GCC diagnostic ignored "-Wformat-truncation" /* Or "-Wformat-truncation" */
`

Wiring is hooked up correctly. I'm using a 3.5mm breakout cable. If I connect the cable to my 3.5mm output on my PC, I can hear sound out of the phone's headset.

But if I connect the 3.5mm breakout to the audio shield, no audio is playing. Not Greeting, not Beep, nothing.

Monitor is showing the proper state changes:

`Serial set up correctly
Mode switched to:  Initialising
Button (pin 1) Release
SD card correctly initialized
Add **SDClass** file system
Added SD card via MTP
Mode switched to:  Ready
&&&&& Dump MTPStorage Loop Data &&&&&
	Callback Data:		0	0x0	0
		1	0xd091	1
		2	0x0	0
		3	0x0	0
		4	0x0	0
	File Systems:
		0	0x200039c8	1
Handset lifted
Mode switched to:  Prompting
Starting Recording
Opened file !
Recording to  00001.wav
Mode switched to:  Recording
Stopping Recording
header written
Subchunk2: 187136
Closed file
Mode switched to:  Ready
Handset lifted
Mode switched to:  Prompting
Button (pin 0) Release
Button (pin 0) Press
Starting Recording
Opened file !
Recording to  00002.wav
Mode switched to:  Recording
Stopping Recording
header written
Subchunk2: 157440
Closed file
Mode switched to:  Ready
`

I get some audio files, but they're all empty. No static or noise.

Anyone running into this issue?

ranasats, 2023-03-23:
The IDE is using the teensy SD library. It's writing files, etc. Just not able to play/record actual sound.

sergioeei, 2023-04-07:
Hi @ranasats, did you found solution meanwhile?

I'm preparing a similar setup. 
For now I just have one Teensy 4.1 board connected to pc to solve compilation erros. 
No other wiring, no Audio Shield and still no sd card.

The program compiles with no errors now.
I didn't get that warning you have.

I got other errors, and the last one was a bit wierd. The line 166 was like this:  void playTone(Tones play);
The compiler was complaining in the 5 places that playTone function was being used Saying "undefined reference to `playTone(Tones)"

So to avoid copilation error I changed line 166 to: void playTone(Tones play) { }

I left the definition of the function empty because I still don't know what to do with it. 
I can antecipate that anything related with tones will not work. I will not listen any tone because the function is empty!! 
Or maybe I'm missing something.

Did you had this problem? 
and if yes how did you solve it? 
did you hear the tones?

Ps.
I'm trying to antecipate problems to get this done in 3 weeks for a weadding.
My programing skills are kind of forgoten but lets dream high :)
So if you solved something that might save me some time please let me know.
any help would be much apreciated.

Also trying to find a quik supplier to portugal for the audio shield. If anyone knows one I'll apreciate the hint.

MichaelWilczynski, 2023-04-07:
I am working on the same thing. However I have run into some other issues. This is all totally new to me and I have zero experience with coding, github or any of this lol. 

I am getting the following error when I try to verify the code. Any help or guidance would be greatly appreciated.

Arduino: 1.8.19 (Windows 10), TD: 1.57, Board: "Teensy 4.1, Serial + MTP Disk (Experimental), 150 MHz, Faster, US English"

Audio_Guest_Book:49: error: 'AudioPlaySdWavX' does not name a type

 AudioPlaySdWavX              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file

 ^

Audio_Guest_Book:54: error: 'playWav1' was not declared in this scope

 AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer

                            ^

Audio_Guest_Book: In function 'void loop()':

Audio_Guest_Book:185: error: 'playWav1' was not declared in this scope

       playWav1.play("greeting");    

       ^

Audio_Guest_Book: In function 'void playAllRecordings()':

Audio_Guest_Book:374: error: 'playWav1' was not declared in this scope

       playWav1.play(entry.name());

       ^

Audio_Guest_Book:380: error: 'playWav1' was not declared in this scope

     while (!playWav1.isStopped()) { // this works for playWav

             ^

Audio_Guest_Book: In function 'void playLastRecording()':

Audio_Guest_Book:411: error: 'playWav1' was not declared in this scope

       playWav1.play(filename);

       ^

'AudioPlaySdWavX' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.