playfultechnology/audio-guestbook #72: Buzzing on recordings

Arrcatype, 2023-03-24:
When ever I playback recordings I only get a buzzing noise with no other audio. I don't know how to fix that

tbn319, 2023-03-24:
I had a similar issue and was able to get the voice recording audio to be louder by adding a ground loop isolator between the TRRS breakout and the Teensy Audio.  However, I am still getting the buzzing sound.  I've tried messing with the micgain as well.  Let me know if the ground loop isolator helps you.

If anything I think it might be the lack of shielding on my handset cord...might try to fix that next.

tbn319, 2023-03-24:
Also, I had to replace my mic for the sound to get better.  I think that helped the most.

Fanie123456, 2023-03-27:
I did go back to the original code from 22.4.2022, when the video was recorded and everything is fine there. seems like writing to wav format is to heavy for the teensy.

Gaunajr, 2023-04-04:
> 

Isnt the original in WAV as well? I had the same issue but added a ground loop isolator and seems to have fixed the problem. Could be something with the wires used for the mic and power right above the mic connection to the teensy.