playfultechnology/audio-guestbook #75: In function 'void continueRecording()

Thisios, 2023-03-25:
C:\Users\Admin\Documents\Arduino\audio-guestbook\audio-guestbook.ino: In function 'void continueRecording()':
C:\Users\Admin\Documents\Arduino\audio-guestbook\audio-guestbook.ino:264:58: error: conflicting declaration 'byte buffer [8192]'
     byte buffer[NBLOX*AUDIO_BLOCK_SAMPLES*sizeof(int16_t)];
                                                          ^
C:\Users\Admin\Documents\Arduino\audio-guestbook\audio-guestbook.ino:263:10: note: previous declaration as 'byte buffer [512]'
     byte buffer[512];
          ^

rthewitt, 2023-04-06:
I presume you've fixed this, but just in case:
Commit faf0838725a21ac5a9a527c7f0aa882bca58df8c removes the first buffer declaration `byte buffer[512];`

The comment on that commit explains that it was a mistake and a duplicate.  The latest version of the code does not have that first line.  I suggest doing a `git pull` to update the code, or a checkout if you're on an older commit.

rthewitt, 2023-04-06:
Excellent!  If possible, please close the issue.

closed 2023-04-06 by Thisios