playfultechnology/audio-guestbook #77: Feature request: Mic feedback or white noise

fuanje, 2023-03-30:
After the greeting and subsequent beep are played the device "feels" like it's turned off even though it's recording just fine.

Is it possible to add minor microphone feedback to the earpiece or some type of white noise when a person is recording their message? So that the device "feels" like it's powered on.

Gaunajr, 2023-04-04:
> You can add a record LED. It will go on when the recording is activated.

Any advice on how to do that? Links to instructions would be helpful as I would like to do that as well. Thanks

zaff0, 2024-03-01:
I created an additional patch cord from the mic (i2s2) to the mixer (input 2) and adjusted that mixer gain low, 0.02f for that input. 
This creates a slight feedback from the mic to the earpiece, like what you get on land line phones.