playfultechnology/audio-guestbook #81: Flipped

SwiFtEnRo, 2023-04-11:
So this code works great (THANK YOU!) I was able to adjust my input volume, my delay timer, and my output volume but I have a weird issue. Mines flipped? Does that mean my cables are backward or is there something I need to change in the phone? When I have the phone "off the hook" it doesn't record but when I press the buttons down reenacting having the phone in the "off" state that's when the greeting starts playing and recording. How do I flip?

SwiFtEnRo, 2023-04-11:
I have tried replacing all "fallingEdge" with "risingEdge" and vice versa but when I do that then nothing works. The phone doesn't work in either state.

SwiFtEnRo, 2023-04-11:
I figured it out!!! So yes switch all variables fallingEdge and risingEdge by using the find and replace all function. Next (and the part why I didn't think it was working) you must change the "hookRead==1" not "0"!!!!! Since its flipped 0 means closed and 1 means open

Correct code:
```
hookRead=digitalRead(HOOK_PIN);  // Check if handset replaced.  
          if (hookRead==1)  {
            Serial.println("Handset Replaced Before Message");
            mode = Mode::Ready;  // Return to ready mode if hang up.
          return;
        }
```

vesebrandon, 2023-04-28:
@SwiFtEnRo You mentioned that you adjusted the output volume.  I'm having an issue where the mic records everything just fine, but the greeting and the tone are EXTREMELY quiet.  

In the code I changed float beep_volume = 0.04f to 1.0, and I was able to hear the tone slightly, but it seems like the overall output volume is low.  Any idea where I should be looking for that?

rogome, 2023-05-15:
where did you put that part of the code? I'm having the same issue but I can't figure it out.

SwiFtEnRo, 2023-05-15:
> 

Good morning @vesebrandon. Sorry, I didn't get back to you after I built my phone I stopped really thinking about it until I saw an email notification that you were having some issues. OK, so you need to up the output volume of the greetings and the tone? The setting I did to do this is:
`sgtl5000_1.volume(0.6);  // Output speaker volume.  default (.5).`
Let me know if changing this value helps you or if you need more help! I'll make sure to have my notifications on.

SwiFtEnRo, 2023-05-15:
> where did you put that part of the code? I'm having the same issue but I can't figure it out.

Umm? Just above? On the same comment, you reacted to with a confused emoji?
What was confusing about it? As quoted above "you must change the "hookRead==1" not "0""

vesebrandon, 2023-05-15:
@rogome you'll be shocked to hear how I fixed my issue... 
I replaced the TRRS cable.

I actually am now using a TRS instead and it's working perfectly.