playfultechnology/audio-guestbook #83: Unable to read the SD Card

Drightiger, 2023-04-15:
I'm not the first to have this problem and I wont be the last; previous threads on this are dead.

Firstly, thanks to Alistair and all the contributors on this project, you are providing a brilliant and frugal project for people to do whilst avoiding the absolute cash cow of the wedding industry (looking at you, At the Beep).

T4.0 w/ Audio shield. Code is running fine and I get the start up tone through the speaker and thereafter abruptly stopped at "Unable to read the SD card". Ive tried:
1. 2 different SD cards of different brands and a further on order, 64GB & 128GB (Both SDXC, could it require SDHC?)
2. Resistance checked the connections between the boards and my solders - all good. [This is a good guide for diagnosing pin connection.](https://forum.pjrc.com/threads/70479-T4-0-audio-shield-SD-problem)
3. Tried the alternative defined pins that has apparently worked for someone else (10, 11, 13 instead of 10, 7, 14, I still don't understand why its the latter but whatever)

So help me god I've put hours into this, if someone helps me with this I'll pay it forward and delve into the YouTube comments for the many people in there struggling and help 1 person finish it off!

Drightiger, 2023-04-25:
Finally got this working, point 2 was my eureka moment:

1. If you are using Audioshield Rev D2 (or Rev D), you NEED to change the pins to 10 11 and 13 instead of 10 7 14. I suspect Alistair was using a Audioshield Rev C in his original code and on the video.
2. I re-soldered my 5V and 3.3v volt pins which were a bit of a mess but I thought they were passable, turns out they were not! 
3. I changed to a SDHC 16GB card as opposed to a SDXC - this might not affect it but it could be potential issue.

If you hear the initial startup tone through the speaker but get the 'Unable to read SD card':

1. Check solders - I didn't want to believe this was my issue as I had experience soldering but it turns out it was. The SD card runs off the 3.3v pin, which is conveniently right next to a Ground pin - make sure they do not overlap (which I suspect was my problem).
2. Resistance check the 10 11 13 pins to make sure they're connected
3. Check your code that it references to 10 11 13 pins!

closed 2023-05-01 by Drightiger

SlephenX, 2023-09-19:
I'm sorry, but why do we have to solder the 3.3V and the GRD pins? I thought they would be put together from attaching the board to the teensy 4.0 and then use the USB power?

Drightiger, 2023-09-19:
You should solder every header to the board, not just the ones that are being used