playfultechnology/audio-guestbook #85: No beep and microphone/speaker not working

RILEYFS13, 2023-04-23:
Hi guys,

I've hit a brick wall with this project. I've wired it all up and coded it and there's no beeps coming through the speakers.

I know the speaker works as if I plug it in to the PC it plays sound. The microphone is also not working however, it's recording (it's just blank files).

I've checked all the wiring and I'm confident it's all wired up correctly. Does anyone have any ideas how to sort this?

rogome, 2023-05-15:
Did you solve it? i have the same problem

RILEYFS13, 2023-05-15:
Not yet, I'm still struggling with it. If I figure it out, I'll let you know. I'd appreciate if you'd do the same.

rogome, 2023-05-15:
I've solved it. Apparently the way I was doing the compilation was wrong. I've added some libraries manually. Most off the info is on the same GitHub, he has made some changes. You have to search his corrections, read everything and follow the instructions, that's all what I can say. I'm sorry I can't say more about it

schaleroo, 2023-08-07:
@RILEYFS13  Did you manage to solve it? I have the same problem, event tried using a new microphone with no avail