playfultechnology/audio-guestbook #89: Multiple Errors while verifying Sketch

Glowey1, 2023-05-10:
Hello, I am very new to this type of thing but have followed all the guides properly and utilized others issues to help solve some of mine however I am still getting issues when I try to Verify the sketch. This happens while I have all 3 of the Audio Guestbook Code open, the play_sd_wav.cpp code open and the play_sd_wav.h code. 

This is the error message coming up, PLEASE Help me get this sorted out!!! Thanks! I have no idea what these are trying to tell me and it is pushing me towards giving up completely.

Arduino: 1.8.19 (Windows 10), TD: 1.58, Board: "Teensy 4.0, Serial + MTP Disk (Experimental), 600 MHz, Faster, US English, Bigger blocks (256 samples)"

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy-main\src/MTP_Storage.h:35,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy-main\src/MTP_Teensy.h:45,

                 from C:\Program Files (x86)\Arduino\Audio_Guestbook_Test\Audio_Guestbook_Test.ino:30:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:37: warning: "FILE_READ" redefined

   37 | #define FILE_READ  0

      | 

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/play_sd_raw.h:32,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/Audio.h:127,

                 from C:\Program Files (x86)\Arduino\Audio_Guestbook_Test\Audio_Guestbook_Test.ino:25:

C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:23: note: this is the location of the previous definition

   23 | #define FILE_READ O_READ

      | 

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy-main\src/MTP_Storage.h:35,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy-main\src/MTP_Teensy.h:45,

                 from C:\Program Files (x86)\Arduino\Audio_Guestbook_Test\Audio_Guestbook_Test.ino:30:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:38: warning: "FILE_WRITE" redefined

   38 | #define FILE_WRITE 1

      | 

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/play_sd_raw.h:32,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/Audio.h:127,

                 from C:\Program Files (x86)\Arduino\Audio_Guestbook_Test\Audio_Guestbook_Test.ino:25:

C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:24: note: this is the location of the previous definition

   24 | #define FILE_WRITE (O_READ | O_WRITE | O_CREAT | O_APPEND)

      | 

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy-main\src/MTP_Storage.h:35,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy-main\src/MTP_Teensy.h:45,

                 from C:\Program Files (x86)\Arduino\Audio_Guestbook_Test\Audio_Guestbook_Test.ino:30:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:75:17: error: reference to 'File' is ambiguous

   75 |         virtual File openNextFile(uint8_t mode=0) = 0;

      |                 ^~~~

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/play_sd_raw.h:32,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/Audio.h:127,

[... 464 lines cut ...]
  439 |   *date = FS_DATE(year(), month(), day());

      |           ^~~~~~~

      |           LS_DATE

Audio_Guestbook_Test:442: error: 'FS_TIME' was not declared in this scope; did you mean 'FAT_TIME'?

  442 |   *time = FS_TIME(hour(), minute(), second());

      |           ^~~~~~~

      |           FAT_TIME

Audio_Guestbook_Test: In function 'void writeOutHeader()':

Audio_Guestbook_Test:470: error: 'frec' was not declared in this scope; did you mean 'free'?

  470 |   frec.seek(0);

      |   ^~~~

      |   free

reference to 'File' is ambiguous

Kenitilas, 2023-05-11:
Hey I am not a pro as well but for me it seems like you have some issues with the standard libraries. I can tell you how I did it:
•	I downloaded the portable version of Arduino so I can separate my projects
	https://downloads.arduino.cc/arduino-1.8.19-windows.zip
•	Then I installed the Teensyduino (Version 1.58) with all modules selected (for the current version there are more needed than shown in the video)
	https://www.pjrc.com/teensy/td_158/TeensyduinoInstall.exe
•	Copy the files “platform.txt” and “boards.local.txt” to “…\hardware\teensy\avr” (wherever your portable Arduino installation is located)
•	Get the MTP_Teensy from (link) and copy the folder to “…\hardware\teensy\avr\libraries” (I usually remove the “-main” from the folder name cause this can lead to issues if you keep it)
	https://github.com/KurtE/MTP_Teensy
•	Your settings for the teensy board seem to be right. I read in the Issues that a CPU Speed of 150 MHz won’t hurt the program but can save a lot of battery

I hope you can solve your problems!

Glowey1, 2023-05-11:
@Kenitilas Thank you, I'll give these things a try. I have taken the code just as it is on Github and the issue appears to be coming from one of the play_sd tabs I have open. Hoping to dive more into it this week and figure it out.

Glowey1, 2023-06-11:
@Kenitilas, I am happy to say that after uninstalling Everything and starting from scratch, I was able to successfully compile the sketch without errors and upload it to my Teensy. I have not yet had a chance to test it with the phone but so far, moving in the right direction!