playfultechnology/audio-guestbook #90: Error compiling board for Teensy 4.0

SJCherry92, 2023-05-13:
Hi all, 

I've done all of the hardware wiring etc and the changes to the txt files for 'board' and 'platform'. All Teensyduino packages were installed and all saved to C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries.

I've copied in the code from the Github library, but i'm now facing the below issue when trying to validate the code and getting it onto the Teensy board itself.

Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Error compiling for board Teensy 4.0.

Any assistance is greatly appreciated.

Kenitilas, 2023-05-15:
Hey SJCherry

> Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD
Not used: C:\Program Files (x86)\Arduino\libraries\SD

Those messages are okay. It just tells you that there are 2 identical libraries on you PC and the compiler is using the Teensy one (this is correct)

Are there any other messages? At the moment there is no indication where the problem could be....
If your problem is consistent try my instructions here: https://github.com/playfultechnology/audio-guestbook/issues/89#issuecomment-1543487599 maybe this can help you

SJCherry92, 2023-05-20:
Thanks Kenitilas,

I've followed your additional step (MTP_Teensy) and saved these into the AVR libraries (screenshot attached). When doing this, I downloaded the file from your Github. This seems to now be triggering a termination. This is the full error code:

Arduino: 1.8.19 (Windows 10), TD: 1.58, Board: "Teensy 4.0, Serial, 600 MHz, Faster, US English"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10819 -build-path C:\Users\XR341MY\AppData\Local\Temp\arduino_build_504454 -warnings=none -build-cache C:\Users\XR341MY\AppData\Local\Temp\arduino_cache_606907 -verbose C:\Users\XR341MY\Audio_Guestbook\Audio_Guestbook.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10819 -build-path C:\Users\XR341MY\AppData\Local\Temp\arduino_build_504454 -warnings=none -build-cache C:\Users\XR341MY\AppData\Local\Temp\arduino_cache_606907 -verbose C:\Users\XR341MY\Audio_Guestbook\Audio_Guestbook.ino

Using board 'teensy40' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr

Using core 'teensy4' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr

Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10819 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "C:\\Users\\XR341MY\\AppData\\Local\\Temp\\arduino_build_504454\\sketch\\Audio_Guestbook.ino.cpp" -o nul

Alternatives for Bounce.h: [Bounce]

ResolveLibrary(Bounce.h)

  -> candidates: [Bounce]

"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10819 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Bounce" "C:\\Users\\XR341MY\\AppData\\Local\\Temp\\arduino_build_504454\\sketch\\Audio_Guestbook.ino.cpp" -o nul

Alternatives for Audio.h: [Audio@1.3]

ResolveLibrary(Audio.h)

  -> candidates: [Audio@1.3]

"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10819 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Bounce" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Audio" "C:\\Users\\XR341MY\\AppData\\Local\\Temp\\arduino_build_504454\\sketch\\Audio_Guestbook.ino.cpp" -o nul

Alternatives for SPI.h: [SPI@1.0]

ResolveLibrary(SPI.h)

  -> candidates: [SPI@1.0]

"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10819 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Bounce" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Audio" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\SPI" "C:\\Users\\XR341MY\\AppData\\Local\\Temp\\arduino_build_504454\\sketch\\Audio_Guestbook.ino.cpp" -o nul

Alternatives for SD.h: [SD@1.2.4 SD@2.0.0]

ResolveLibrary(SD.h)

  -> candidates: [SD@1.2.4 SD@2.0.0]

"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5- [... cut ...]
Using library SdFat at version 2.1.2 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SdFat 

Using library SerialFlash at version 0.5 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SerialFlash 

Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 

Using library Time at version 1.6.1 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Time 

Error compiling for board Teensy 4.0.

Any help greatly appreciated.

[image]

TWK-Nattttttttt, 2023-05-26:
I'm having this same issue! Were you ever able to find a solution?

SJCherry92, 2023-05-26:
I haven’t. If anyone on the community has suggested fixes that would be fantastic

Kenitilas, 2023-05-26:
> C:\Users\XR341MY\Audio_Guestbook\Audio_Guestbook.ino:30:10: fatal error: MTP_Teensy.h: No such file or directoryResolveLibrary(MTP_Teensy.h)
-> candidates: [] 30 | #include <MTP_Teensy.h>

You still have problems including the MTP library. What I did was downloaded the MTP Folder, extracted it and you get a folder with "MTP_Teensy-main" just remove the "-main" part of the folder name to "MTP_Teensy" and copy this folder with everything in it to “…\hardware\teensy\avr\libraries”

I tried to visualize this in 2 pictures

[image]

SJCherry92, 2023-05-26:
Thanks, this has now worked and i've been able to verify and then upload the code (code below if helpful for anyone who sees this). I also had to change audiopaysdwavX to audiopaysdwav in the code.

@Kenitilas I think the code is now all fine, but when I have the Teensy connected to the phone and connected to a battery pack I just hear a dull noise (that then stops when I press the switch on the phone. Any ideas of final troubleshooting?

/**
 * Audio Guestbook, Copyright (c) 2022 Playful Technology
 * 
 * Tested using a Teensy 4.0 with Teensy Audio Shield, although should work 
 * with minor modifications on other similar hardware
 * 
 * When handset is lifted, a pre-recorded greeting message is played, followed by a tone.
 * Then, recording starts, and continues until the handset is replaced.
 * Playback button allows all messages currently saved on SD card through earpiece 
 * 
 * Files are saved on SD card as 44.1kHz, 16-bit, mono signed integer RAW audio format 
 * --> changed this to WAV recording, DD4WH 2022_07_31
 * --> added MTP support, which enables copying WAV files from the SD card via the USB connection, DD4WH 2022_08_01
 * 
 * 
 * Frank DD4WH, August 1st 2022 
 * for a DBP 611 telephone (closed contact when handheld is lifted) & with recording to WAV file
 * contact for switch button 0 is closed when handheld is lifted
 * 
 * GNU GPL v3.0 license
 * 
 */

#include <Bounce.h>
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <TimeLib.h>
#include <MTP_Teensy.h>
#include "play_sd_wav.h" // local copy with fixes

// DEFINES
// Define pins used by Teensy Audio Shield
#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  7
#define SDCARD_SCK_PIN   14
// And those used for inputs
#define HOOK_PIN 0
#define PLAYBACK_BUTTON_PIN 1

#define noINSTRUMENT_SD_WRITE

// GLOBALS
// Audio initialisation code can be generated using the GUI interface at https://www.pjrc.com/teensy/gui/
// Inputs
AudioSynthWaveform          waveform1; // To create the "beep" sfx
AudioInputI2S               i2s2; // I2S input from microphone on audio shield
AudioPlaySdWav              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file
AudioRecordQueue            queue1; // Creating an audio buffer in memory before saving to SD
AudioMixer4                 mixer; // Allows merging several inputs to same output
AudioOutputI2S              i2s1; // I2S interface to Speaker/Line Out on Audio shield
AudioConnection patchCord1(waveform1, 0, mixer, 0); // wave to mixer 
AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer
AudioConnection patchCord4(mixer, 0, i2s1, 0); // mixer output to speaker (L)
AudioConnection patchCord6(mixer, 0, i2s1, 1); // mixer output to speaker (R)
AudioConnection patchCord5(i2s2, 0, queue1, 0); // mic input to queue (L)
AudioControlSGTL5000      sgtl5000_1;

// Filename to save audio recording on SD card
char filename[15];
// The file object itself
File frec;

// Use long 40ms debounce time on both switches
Bounce buttonRecord = Bounce(HOOK_PIN, 40);
[... 450 lines cut ...]
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
}

void print_mode(void) { // only for debugging
  Serial.print("Mode switched to: ");
  // Initialising, Ready, Prompting, Recording, Playing
  if(mode == Mode::Ready)           Serial.println(" Ready");
  else if(mode == Mode::Prompting)  Serial.println(" Prompting");
  else if(mode == Mode::Recording)  Serial.println(" Recording");
  else if(mode == Mode::Playing)    Serial.println(" Playing");
  else if(mode == Mode::Initialising)  Serial.println(" Initialising");
  else Serial.println(" Undefined");
}

Kenitilas, 2023-05-26:
I had this issue when my "greeting.wav" was in the wrong format. ATM I am at work so I can't tell you my settings for audacity. At first I used a recorded message by the phone and renamed it on the PC to "greeting.wav" those files are always in the correct format.

SJCherry92, 2023-05-26:
Thanks @Kenitilas. Audio file on the SD all looks to be fine and in WAV. Still facing the same issue though. Could it be related to hardware wiring?

Kenitilas, 2023-05-27:
When exporting the wav file be sure you have 44100 Hz Sample Rate 
[image]

And for the encoding I am not sure anymore but I think I used "Signed 16-bit PCM", but if you encouter the same problem you should try some other.
Your recording should sound the best if you have a mono channel.

If you're not shure about your wiring just share some pictures

SJCherry92, 2023-05-28:
Thanks, I’ve set the right rate but still facing the same issue. Wiring pictures attached

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Kenitilas, 2023-05-30:
On a first glance your connections seems to be flimsy but this shouldn’t be the problem right now since you can head something but your message and since you are not using a playback button I don’t see any potential ground loop. I assume you double checked all your connections with a multimeter already.
If I don’t forget it this evening I will try to record a message with my phone and share it to you. This message should be playable!