playfultechnology/audio-guestbook #91: Never stop recording

rogome, 2023-05-15:
I made my audio guestbook from an aliexpress "antique" phone. As shown in the video, my hanging button is backward. (Like the old office phone he checks). I've changed all the"button_recording.risingEdge()" for "button_recording.fallinEdge()" and vice versa. when I hang my phone, it beeps a couple of times, then It records for a second and starts again. It does the same even if you don't change the ringingEdge for fallingEdge. Does anyone know how to solve it? Is it supposed to work that way? 
I'm adding a copy of the code version that I'm using.

[AudioGuestbook.txt](https://github.com/playfultechnology/audio-guestbook/files/11473988/AudioGuestbook.txt)

SwiFtEnRo, 2023-05-15:
@rogome Good morning,
Looking over your code here and where did you get this from? Your problem is in the "case Mode::Prompting:" stage.
I don't have too much time to look at this since I have to get to work so I'll reply later but where is your counter? Like how are you incrementing the messages and if the phone is replaced getting rid of that number in the count since its no longer relevant? 
Here is what my "Prompting" stage code looks like:
`case Mode::Prompting:

    
             // Wait a second for users to put the handset to their ear.   RAM
             // Added to read hook pin to check if handset has been replaced before proceeding.  RAM
                      
      x=0;   // Clear counter.
      while (x < 30)  {
        wait(50);   // Wait 50 mS.  
        x=x+1;      // Increment counter
        Serial.println(x);  // Print counter value
        hookRead=digitalRead(HOOK_PIN);  // Check if handset replaced.  
          if (hookRead==1)  {
            Serial.println("Handset Replaced Before Message");
            mode = Mode::Ready;  // Return to ready mode if hang up.
          return;
        }
      }
      
 
           
      // Play the greeting inviting them to record their message
      playRaw1.play("greeting.raw");    //  Changed from default (.wav) to (.raw) format.  Greeting MUST be recorded in .raw format.   RAM
      // Wait until the  message has finished playing
      while (playRaw1.isPlaying()) {
      
        // Check whether the handset is replaced
        buttonRecord.update();
        // Handset is replaced
        if(buttonRecord.risingEdge()) {
          playRaw1.stop();
          mode = Mode::Ready;
          return;

          
        }
        
      }
`

rogome, 2023-05-15:
@SwiFtEnRo I took the code from here: https://github.com/playfultechnology/audio-guestbook/blob/main/audio-guestbook.ino
It is probably a different version.
I know a little bit of coding but I'm mostly an enthusiasts.
I saw the version that u are using on a different forum, but didn't worked for me...

rogome, 2023-05-16:
@SwiFtEnRo the coda was updated, your version doesn't work for me.  
In my version he is using   
// Configure the input pins
  pinMode(HOOK_PIN, INPUT_PULLUP);
  pinMode(PLAYBACK_BUTTON_PIN, INPUT_PULLUP);
After what I've been reading the input_pullup is reading if the button is pressed or if it is not. If I change it for input_pulldown and it is not working better. 
Any idea?

rogome, 2023-05-16:
Just in case someone reads this in the future.  If you have a flipped phone like mine and You are using the updated version of the code, You have to swap all the fallingEdge for risingEdge for the button record (and vice-versa).
pinMode(HOOK_PIN, INPUT_PULLUP);
pinMode(PLAYBACK_BUTTON_PIN, INPUT_PULLUP);
next step would be changing the pullup for pulldown
and then you have to change the hook connection from GND to 3.3V.
This is very important. If you do that, you'll have to add a resistor, otherwise, you will kill the teensy board.
This part depends on your phone hook button. Please double-check it.

closed 2023-05-16 by rogome