playfultechnology/audio-guestbook #92: Unable to access the SD card

Jipsey-A, 2023-05-20:
Hi!
I'm fairly confident in my hardware installation - I've stumbled my way through the software side of things. I think I've done the new set-up correctly (although "Audio tweaks" does not appear for me).  My teensy 4.0 appears to have accepted the code, but the serial read says 'Unable to access SD card'. My SD card was empty, I loaded a Greeting.wav file recorded on Audacity saved as 16 bit, 44.1kHz. I put it into the teensy set-up and received the 'unable to access SD card'. When I put it back into my laptop I can now see a second file called mtpindex.dat 
I wonder whether it's something to do with the way the SD card's formatted?? 

Grateful for your help!

Jipsey-A, 2023-05-21:
After reading other issues on here I purchased a new 'A1' SD card, I reformatted this to 'exFAT' and the serial monitor now shows:
19:54:02.605 -> Serial set up correctly
19:54:02.605 -> Audio block set to 128 samples
19:54:02.605 -> Mode switched to:  Initialising
But it doesn't do anything else...

Kenitilas, 2023-05-26:
I am not sure if you still have a SD Card problem. I usually use SD Memory Card Formatter to prepare my cards 
[https://www.sdcard.org/downloads/formatter/](url)

Do you want the 128 sample blocks or is this an issue with the bigger blocks?
I had a ground loop in my wiring and thus couldn't hear or record anything

Jipsey-A, 2023-05-26:
Hi @Kenitilas 
I got the SD card working but now (depending on which way I code falling and rising edge) I either find that the hook switch works in reverse, or that I get a false hang-up a moment into the greeting message. I think this is caused by an issue with my wiring the circuit board. Please could you share where / how you found the ground loop? I suspect this may be the problem for me too!
Thank you!

Kenitilas, 2023-05-26:
Hey 
keep in mind that the switch of the handle button is multiple times in the code! I changed the code to the wrong setup on purpose, because this way it was way easier for me testing the phone by just tapping the handle, but maybe this is just my weird thing ;)
I found my ground loop by taking the multimeter and measuring all wires by its own. This may take some time but so I found a wrong-wired button. In my case I wired the ground level to an input pin of the teensy. You can also check the phone for a universal ground. Mine already had big ground trace on the back of the PCB I could use for all my GND connections.
Hope this helps!