playfultechnology/audio-guestbook #95: Help Me & Get a Coffee! | Teensy 4.1 w/ Audio Shield Rev D no audio/recording/playback

Bean-17, 2023-05-31:
Hello! You read that right, I am offering a coffee (I'll Venmo/PayPal you some money) for whoever helps me solve the issue(s) I am having :) 

As stated I have a Teensy 4.1 board, with the audio shield (Revision D), and I am using a Western Electric Model 500 Rotary phone. Please see the google drive folder link for images of my current set up and screen grabs of code.

[https://drive.google.com/drive/folders/1LZMS9A4jORcyJqAjhnyBZ26KTtIE16Dk?usp=drive_link](https://drive.google.com/drive/folders/1LZMS9A4jORcyJqAjhnyBZ26KTtIE16Dk?usp=drive_link)

I have already done a good bit of trouble shooting and have arrived at a point where I am close but yet so far. Some things to note:

• There are no current errors or warnings when compiling or uploading. 
• Board is getting power (I have checked the 3V and 5V pins with the GND and they are reading out correctly)
• Made all the adjustments mentioned in the OG post, and have done the MTP library add with that "Kurt" thing. 
• Headset seems to work independent from the Teensy. By this I mean I can visit "https://mictests.com/" and when I plug the headphone jack directly into the laptop instead of the audio shield, I can both record and hear the audio on playback through the phone headset. The switch also works with this to turn off the audio playback. Additionally, whit this same set up I can use the phone headset to listen to other audio (videos/music) and the switch will "turn off" the audio with pressed. 
• Finally,  as you can see in the image of the code in the Google Drive folder, I have switched to the Built In SD card. I figured this would not work, but as you can see from the image of my serial monitor report, it seems to be finding and writing to the SD card just fine. 

So I am at the point where I can tell the wiring is correct (potentially a grounding issue though?), the serial monitor is sending the correct messages, files are being created on the SD card, and the board is getting power. 

I do not know what to do past everything else I have already looked into, so all help is appreciated! 

Thank you all in advance.

Kenitilas, 2023-06-05:
I think your problem could be in using a teensy 4.1 while the code was made for the teensy 4.0. Especially the second SD card holder can cause a problem when the controller tries to use the wrong one.
My build included a teensy 4.0 so I don’t have any experience with the 4.1 controller but have a look at this fork: https://github.com/DD4WH/audio-guestbook

owelchez, 2023-06-15:
Hello,
I would help you for free if I could, I also did this project with a Rotary Phone and Teensy 4.0 but I used DD4WH's Fork, even though he mentions that his code doesn't work, it worked flawlessly on my phone. I got my Teensy at a local store called Micro Center.

aaronwbrown, 2023-06-18:
@Bean-17 - were you able to get this solved? I am about to jump into this with a 4.1, and curious if you've found any gotchas that others attempting should look out for. 

As I go through my setup, I'll share any learnings here as well. 👍

aaronwbrown, 2023-06-27:
@Bean-17 - A couple things I learned during the build:

- Stacking the headers without soldering them will not work - I'm new to hardware and I learned this the hard way. The connections need to be super well established, and simply connecting does not provide for a tight enough connection. 
- Since you're using a 4.1, you need to modify this [line](https://github.com/playfultechnology/audio-guestbook/blob/main/audio-guestbook.ino#L128C21-L128C21) to this:
```
if (!(SD.begin(BUILTIN_SDCARD))) 
```

Do the above and you should have functional audio recording and playback. 

What's still not working for me is the greeting. There's a lot of commented out code in `audio-guestbook.ino`, so I think it's a matter of doing some more read through to get it working. Let me know if you have any luck, and I'll do the same.

aaronwbrown, 2023-07-04:
@Bean-17 - I used Quicktime to record the audio, but didn't check the rate. Even though I read through the README, where it [explicitly mentions](https://github.com/playfultechnology/audio-guestbook/blob/main/README.md?plain=1#L53) that the greeting must be recorded at a rate of 44.1ksps. Once I converted it to 44.1ksps, the greeting worked just fine.  

If you're on a *nix and have `ffmpeg` installed, you can use this line to change the rate. 

`ffmpeg -i input.wav -ar 44100 output.wav`

I have a [forked version of the code](https://github.com/aaronwbrown/audio-guestbook_4.1) that worked for my Teensy 4.1. Hopefully you can get it to work for you, too.