playfultechnology/audio-guestbook #99: Compiling Issues on Mac 13.1 - Arduino 2.1.0

sterndust, 2023-06-15:
Hi all!

First off thank you all for the tutorials and answering all the issues, this is my first project and I'm pretty savvy but I'm having some similar issues that others have reported, but the solutions haven't been able to help me, and the only difference I am seeing is that I'm running Arduino IDE 2.1.0 for Mac (os 13.1)

• I followed the tutorial on pjrc to install Teensyduino 1.5.8 add-on (https://www.pjrc.com/teensy/td_download.html)

• I copied the modified Arduino architecture configuration files "platform.txt" and "boards.txt" into the proper directory for OSX 
/Users/$username/Library/Arduino15/packages/teensy/hardware/avr/1.5.8
When I launched Arduino, the option for "Audio Tweaks" now shows up, so I'm pretty confident that worked.

• I downloaded the MTP_Teensy.zip and removed the "-main" and copied the folder: /Users/$username/Library/Arduino15/packages/teensy/hardware/avr/1.58.1/libraries

• I changed USB Type to MTP Disk (Experimental)

When I try to compile, I get the following error
- - - - - - - - 
fork/exec /Users/dylanstern/Library/Arduino15/packages/teensy/hardware/avr/../tools/arm/bin/arm-none-eabi-g++: no such file or directory

Compilation error: fork/exec /Users/dylanstern/Library/Arduino15/packages/teensy/hardware/avr/../tools/arm/bin/arm-none-eabi-g++: no such file or directory
- - - - - - - - 

Any help would be greatly appreciated!

mitchell-massey, 2023-06-15:
Same error here.

SlephenX, 2023-09-19:
Were you able to find anything?