playfultechnology/audio-guestbook #102: Audio Playback Fast/Slow

Wucainclan, 2023-06-21:
Hey All,

I'm using a Teensy 4.0 with a Audio Shield Rev D2 and am experiencing some issues with playing a greeting message/recordings. I've verified all connections are solid and even tested the speaker/mic combo. I have my SD card (Sandisk Ultra) plugged directly into the Audio Shield and it's been formatted (I've also tried on multiple SD cards to no avail). Please reference my conversation [here](https://github.com/playfultechnology/audio-guestbook/issues/67)

The issue I'm having is the greeting message sounds slow and robotic (like it's playing at .5 speed) and the recorded messages are fast and robotic (like it's playing at 2x speed). Any ideas? It's strange to me that it's happening on both a pre-recorded greeting, as well as any recordings the device makes.

closed 2023-07-03 by Wucainclan

Jeffff34, 2023-07-15:
Hello, Do you find an issu to your problem?
I have the same problem and I don't know how fix it.
Thank you

Jeffff34, 2023-07-16:
I find the solution for the greeting message. For me it's the format of th audio. 
I saved in wave 44,1 but 32 Bit not 16.
Greeting message is Ok now