playfultechnology/audio-guestbook #106: Unable to access the SD card - teensy 4.1

vatalla30, 2023-07-16:
Ive already created one using the Teensy 4.1 and it worked just fine with the original code and just changing the pin numbers but I cant for the life of me figure out what the issue is this time. Im pretty confident in my soldering (but ill add pics here) and the code is uploading and verfying fine but the serial monitor says "Unable to access SD card" and nothing happens when I connect the phone to the battery pack and power on. 
[image]
[image]

GazzaH1, 2023-09-19:
Hi I having the exact same issue and I have no idea what to do. Same as in the code is validating and seem to be uploading into the teensy correctly but when running serial monitor keeps repeating "Unable to access SD card" over and over. I'm absolutely stumped.

gunnercooke, 2023-09-20:
I have the same issue.

bvbyrd, 2023-12-20:
Would one of you mind Sharing the code you used for this project on 4.1?  I'm missing something on my end.

Thanks

laurentaylor333, 2025-07-20:
Has anyone found a resolution to the SD card issue? I am having the same issues.