playfultechnology/audio-guestbook #111: No sound in recordings

relishandpickles, 2023-07-23:
I've been messing around with this thing for months. 

A lot of troubleshooting later, and everything seems to work but the recordings. Telephone beeps, plays the greeting, goes into the recording phase, lights flash on the device showing that it is recording, hang up the phone and it stops as it should. 

The problem is that when I access the recordings, there is no sound. The files seem to contain data proportional to how long the recording was, but I don't hear anything. Audacity shows there is no waveforms in the recording. 

I've reinstalled everything from the ground up over and over. No compiling errors. I've soldered over and over, checked all the connections, and cannot for the life of me figure out why the recordings have no sound. 

Am I missing something? Any idea of why everything would work besides it capturing audio?

schneekluth, 2023-07-31:
Same here. I have checked wiring and soldering and my recordings only contain some kind of ticking noises.

schneekluth, 2023-08-01:
@relishandpickles Hey, I have just figured out that the microphone in the old telephone I am using is simply broken. Maybe you did everything right like me but the old hardware is the issue.

schaleroo, 2023-08-07:
Did you solve it? I have the same probem, even tried a new working microphone with no results

schneekluth, 2023-08-07:
I have indeed solved it an it was **not** the microphone. I had to break a second circuit directly on the pcb of the phone. 

**Try the following:** Search a microphone with 3.5mm headphone jack (e.g. old PS4 headset). The position of the 3.5mm jack rings for thre microphone and ground have the same distance as the mic connectors on the audio shield. Place the jack's rings directly against the pins and try to record some audio. If this works your problem is not the microphone.

DianeM8739, 2023-08-27:
im still having this issue. maybe its my hardware connections? I also had to reverse the falling and risingedge because of the way my phone is setup. I hear the greeting fine but just empty recordings.

DianeM8739, 2023-08-27:
Maybe i messed up the code?

`/**
 * Audio Guestbook, Copyright (c) 2022 Playful Technology
 * 
 * Tested using a Teensy 4.0 with Teensy Audio Shield, although should work 
 * with minor modifications on other similar hardware
 * 
 * When handset is lifted, a pre-recorded greeting message is played, followed by a tone.
 * Then, recording starts, and continues until the handset is replaced.
 * Playback button allows all messages currently saved on SD card through earpiece 
 * 
 * Files are saved on SD card as 44.1kHz, 16-bit, mono signed integer RAW audio format 
 * --> changed this to WAV recording, DD4WH 2022_07_31
 * --> added MTP support, which enables copying WAV files from the SD card via the USB connection, DD4WH 2022_08_01
 * 
 * 
 * Frank DD4WH, August 1st 2022 
 * for a DBP 611 telephone (closed contact when handheld is lifted) & with recording to WAV file
 * contact for switch button 0 is closed when handheld is lifted
 * 
 * GNU GPL v3.0 license
 * 
 */

#include <Bounce.h>
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <TimeLib.h>
#include <MTP_Teensy.h>
#include "play_sd_wav.h" // local copy with fixes

// DEFINES
// Define pins used by Teensy Audio Shield
#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  7
#define SDCARD_SCK_PIN   14
// And those used for inputs
#define HOOK_PIN 0
#define PLAYBACK_BUTTON_PIN 1

#define noINSTRUMENT_SD_WRITE

// GLOBALS
// Audio initialisation code can be generated using the GUI interface at https://www.pjrc.com/teensy/gui/
// Inputs
AudioSynthWaveform          waveform1; // To create the "beep" sfx
AudioInputI2S               i2s2; // I2S input from microphone on audio shield
AudioPlaySdWavX              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file
AudioRecordQueue            queue1; // Creating an audio buffer in memory before saving to SD
AudioMixer4                 mixer; // Allows merging several inputs to same output
AudioOutputI2S              i2s1; // I2S interface to Speaker/Line Out on Audio shield
AudioConnection patchCord1(waveform1, 0, mixer, 0); // wave to mixer 
AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer
AudioConnection patchCord4(mixer, 0, i2s1, 0); // mixer output to speaker (L)
AudioConnection patchCord6(mixer, 0, i2s1, 1); // mixer output to speaker (R)
AudioConnection patchCord5(i2s2, 0, queue1, 0); // mic input to queue (L)
AudioControlSGTL5000      sgtl5000_1;

// Filename to save audio recording on SD card
char filename[15];
// The file object itself
File frec;

// Use long 40ms debounce time on both switches
Bounce buttonRecord = Bounce(HOOK_PIN, 40);
Bounce buttonPlay = Bounce(PLAYBACK_BUTTON_PIN, 40);

[... 448 lines cut ...]
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
}

void print_mode(void) { // only for debugging
  Serial.print("Mode switched to: ");
  // Initialising, Ready, Prompting, Recording, Playing
  if(mode == Mode::Ready)           Serial.println(" Ready");
  else if(mode == Mode::Prompting)  Serial.println(" Prompting");
  else if(mode == Mode::Recording)  Serial.println(" Recording");
  else if(mode == Mode::Playing)    Serial.println(" Playing");
  else if(mode == Mode::Initialising)  Serial.println(" Initialising");
  else Serial.println(" Undefined");
}`

BGOLDEN1994, 2023-11-16:
I had this same issue. I chased code, replaced wires, replace the mic, everything you can think of. 
Turns out it was how the phone was designed. 

When first building the phone, I found the microphone pins on the circuit board itself and soldered directly to those WITHOUT disconnecting the mic from the board, similar to how he does in the video. This was a mistake, since the microphone has some sort of transformer in parallel with the mic which throws off the teensy, probably by drawing lots of power from the mic output. 

To fix this, I took a Dremel and cut all of the board traces that also ran to the handset mic. If I were to do this again, I would make sure to run a direct connection from the handset mic to the teensy by cutting and splicing directly to the handset mic wires.