playfultechnology/audio-guestbook #113: Recording while Phone down

saladtable, 2023-07-25:
I have everything working at the moment, apart from the phone is recording when it is placed on the holder, then stops recording when I pick the phone up. How do I swap this round?

dallingilbert, 2023-07-25:
Not sure how your switch for the phone operates, but if you are looking to change this behavior look into the loop() function on line 163. Depending on what you want, you need to change the behavior from fallingEdge() to risingEdge() on the respective lines.

swoite, 2023-10-08:
I managed to solve this issue by changing line 193 in audio-guestbook.ino. 

Under Mode::Prompting

`    case Mode::Prompting:
      // Wait a second for users to put the handset to their ear
      wait(1000);
      // Play the greeting inviting them to record their message
      playWav1.play("greeting.wav");    
      // Wait until the  message has finished playing
//      while (playWav1.isPlaying()) {
      while (!playWav1.isStopped()) {
        // Check whether the handset is replaced
        buttonRecord.update();
        buttonPlay.update();
        // Handset is replaced
        if(buttonRecord.risingEdge()) {
          playWav1.stop();
          mode = Mode::Ready; print_mode();
          return;
        }
        if(buttonPlay.fallingEdge()) {
          playWav1.stop();
          //playAllRecordings();
          playLastRecording();
          return;
        }
        
      }
`

Change `if(buttonRecord.risingEdge())` to `if(buttonRecord.read() == 1)`

I believe what happens is if the handset is replaced during the `wait(1000);`, the falling edge isn't caught and it just continues to do its thing until the next rising edge occurs. With this change, if the handset is placed down it will stop recording no matter what.