playfultechnology/audio-guestbook #114: Audio "Tweaks" tab not appearing under 4.0

dallingilbert, 2023-07-26:
As stated in the title, I edited the boards.txt file to include the menu options for both 4.0 and 4.1 teensy boards. Unfortunately I can get the audio tweaks tab to appear under the "Tools" menu under only the 4.1 teensy board. When I switch to the 4.0 version of teensy, that option goes away.

gastentwickler, 2023-08-03:
I have the same problem the other way round: I have the option only for the 4.0 teensy board and not for the 4.1

tylerhovi, 2023-08-08:
Also having this issue getting the the Audio Tweak menu to appear.  Using a 4.0 board.

damase38, 2023-08-11:
I'm also having this issue. 4.0 board and doesn't appear.
Any solutions?

RowanZee18, 2023-09-08:
Anyone find any solutions for this. I have tried various things with no luck

EWeezyPeezy, 2023-10-10:
Had the same issue.  This was the fix:

Upon restart, a [known bug in Arduino IDE 2.0](https://github.com/arduino/arduino-ide/issues/1030) may prevent it from recognizing changes in the boards.txt and boards.local.txt files. If you encounter this issue, delete the C:\Users\<user name>\AppData\Roaming\arduino-ide\ folder for the "Audio Tweaks" to appear correctly.

Stevesgoodcoding, 2024-05-06:
EWeezyPeezy's comment above about deleting the file worked for me. Audio settings now show up. Thank you!