playfultechnology/audio-guestbook #116: SD card is not connected or unusable :-(

DianeM8739, 2023-08-27:
Hi i need assistance. I tried using the serial monitor after the code was uploaded and I kept getting the error UNABLE TO ACCESS SD CARD. ran the SD test and got this error: SD card is not connected or unusable :-(

Card is already formatted. I used https://www.sdcard.org/downloads/formatter/sd-memory-card-formatter-for-windows-download/
I've used different SD cards and keep getting the same error

Am i missing something? How can i fix this?

DianeM8739, 2023-08-28:
Nevermind I got it!

bryandeagle, 2023-09-01:
@DianeM8739 Can you elaborate on what you had wrong? I'm experiencing a similar issue.

Vinestou, 2023-09-02:
Same here! Please let us know @DianeM8739

bryandeagle, 2023-09-03:
I solved my issue. I didn’t realize the ground pins on the audio shield are not connected to each other, and they all need to be connected to the teensy for the SD card to work

Vinestou, 2023-09-03:
> I solved my issue. I didn’t realize the ground pins on the audio shield are not connected to each other, and they all need to be connected to the teensy for the SD card to work

Thanks for you reply. Does that include the gnd from the mic? I have built the board without connecting it to the Phone. maby that is my problem (no GND connected).

Vinestou, 2023-09-03:
@bryandeagle Ok soldered everything and the SD can be found!

DianeM8739, 2023-09-03:
That was the same issue i had. Thanks.

cp1797, 2023-10-07:
I have a Teensy 4.0 with Audio shield Rev D2 hooked up with soldered on headers. I've tried everything but the SD Card just doesn't detect. Keeps showing "Unable to access the SD card".

Going by [this comment](https://github.com/playfultechnology/audio-guestbook/issues/116#issuecomment-1703992567), I've checked with a multimeter the connection of the G pin between 5v and 3.3v and the GND pin and they are connected. CS is set to 10. I've also tried 7 through 14.

What am I doing wrong?? Would really appreciate the help.

jogiblade, 2023-11-22:
> I solved my issue. I didn’t realize the ground pins on the audio shield are not connected to each other, and they all need to be connected to the teensy for the SD card to work

Hello,
which Pins have you soldered? Can you please specify?

bryandeagle, 2023-11-22:
There are two ground pins on the audio shield. They both must be connected to the teensy to function.