playfultechnology/audio-guestbook #117: Newbie trying to make this work

PyrotekFX, 2023-08-28:
Hello, sorry if this is a really silly question, I'm trying to make this work and the MTP library keeps coming up as missing, is there a way to download a compiled library or can someone point me in the direction of a video which I can follow step by step to get this done? Thank you.

PyrotekFX, 2023-08-28:
Got the MTP working, everything seems to be working ok, all except I get a clicking in the audio when I'm listening to the recording and when I make my recording its there too. The SD card is a Sandisk Ultra so should be fast enough, dont know if I'm missing something. File of the audio has been uploaded to see if anyone has a fix. I tried doing the 256 sample thing and it just make the greeting super slow.

/**
 * Audio Guestbook, Copyright (c) 2022 Playful Technology
 * 
 * Tested using a Teensy 4.0 with Teensy Audio Shield, although should work 
 * with minor modifications on other similar hardware
 * 
 * When handset is lifted, a pre-recorded greeting message is played, followed by a tone.
 * Then, recording starts, and continues until the handset is replaced.
 * Playback button allows all messages currently saved on SD card through earpiece 
 * 
 * Files are saved on SD card as 44.1kHz, 16-bit, mono signed integer RAW audio format 
 * --> changed this to WAV recording, DD4WH 2022_07_31
 * --> added MTP support, which enables copying WAV files from the SD card via the USB connection, DD4WH 2022_08_01
 * 
 * 
 * Frank DD4WH, August 1st 2022 
 * for a DBP 611 telephone (closed contact when handheld is lifted) & with recording to WAV file
 * contact for switch button 0 is closed when handheld is lifted
 * 
 * GNU GPL v3.0 license
 * 
 */

#include <Bounce.h>
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <TimeLib.h>
#include <MTP_Teensy.h>
#include "play_sd_wav.h" // local copy with fixes

// DEFINES
// Define pins used by Teensy Audio Shield
#define SDCARD_CS_PIN    10
#define SDCARD_MOSI_PIN  7
#define SDCARD_SCK_PIN   14
// And those used for inputs
#define HOOK_PIN 0
#define PLAYBACK_BUTTON_PIN 1

#define noINSTRUMENT_SD_WRITE

// GLOBALS
// Audio initialisation code can be generated using the GUI interface at https://www.pjrc.com/teensy/gui/
// Inputs
AudioSynthWaveform          waveform1; // To create the "beep" sfx
AudioInputI2S               i2s2; // I2S input from microphone on audio shield
AudioPlaySdWav              playWav1; // Play 44.1kHz 16-bit PCM greeting WAV file
AudioRecordQueue            queue1; // Creating an audio buffer in memory before saving to SD
AudioMixer4                 mixer; // Allows merging several inputs to same output
AudioOutputI2S              i2s1; // I2S interface to Speaker/Line Out on Audio shield
AudioConnection patchCord1(waveform1, 0, mixer, 0); // wave to mixer 
AudioConnection patchCord3(playWav1, 0, mixer, 1); // wav file playback mixer
AudioConnection patchCord4(mixer, 0, i2s1, 0); // mixer output to speaker (L)
AudioConnection patchCord6(mixer, 0, i2s1, 1); // mixer output to speaker (R)
AudioConnection patchCord5(i2s2, 0, queue1, 0); // mic input to queue (L)
AudioControlSGTL5000      sgtl5000_1;

// Filename to save audio recording on SD card
char filename[15];
// The file object itself
File frec;

// Use long 40ms debounce time on both switches
Bounce buttonRecord = Bounce(HOOK_PIN, 40);
Bounce buttonPlay = Bounce(PLAYBACK_BUTTON_PIN, 40);

[... 449 lines cut ...]
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
        wait(250);
        waveform1.amplitude(beep_volume);
        wait(250);
        waveform1.amplitude(0);
}

void print_mode(void) { // only for debugging
  Serial.print("Mode switched to: ");
  // Initialising, Ready, Prompting, Recording, Playing
  if(mode == Mode::Ready)           Serial.println(" Ready");
  else if(mode == Mode::Prompting)  Serial.println(" Prompting");
  else if(mode == Mode::Recording)  Serial.println(" Recording");
  else if(mode == Mode::Playing)    Serial.println(" Playing");
  else if(mode == Mode::Initialising)  Serial.println(" Initialising");
  else Serial.println(" Undefined");
}
[00000.zip](https://github.com/playfultechnology/audio-guestbook/files/12454807/00000.zip)

DianeM8739, 2023-08-28:
Can you send me a pic of the wiring. I got far into building it but i can’t for the life of me figure out why it’s not picking up any audio for my recordings. Also might be able to help your ticking problem if i can take a look because mine was doing that too.

PyrotekFX, 2023-08-28:
2 white cables are joined to grey and purple are the hang up switch, yellow and black are connected to white and black are the mic, red and green are connected to speaker cable, I'll be attaching everything to a terminal block once it's ready to be closed up and put together. I didn't make a button for play all recordings because it's not something I want guests to be able to do.
[image]

PyrotekFX, 2023-08-28:
[image]
[image]
[image]

DianeM8739, 2023-08-28:
Omg! Are we working on the same phone??? Is that the Crosley pay phone??? 
Ok so for the rec i connected it to the R on the output side of audio shield and the other one right next to it G. Since i didn’t put in the audio Jack. I can send a pic in a few.

PyrotekFX, 2023-08-29:
I think that might be your problem, the coding might be written in a way to make it route through the 3.5mm port rather than the pin terminals, I just went to the electronics store and bought a mono 3.5mm cable to mono 3.5mm, cut it in half and spliced the cables together with the phones and it worked.

My phone is the knock off Crosley pay phone I found on Alibaba. In Australia the proper Crosley pay phone is astronomically expensive because no one has it here in stock.

DianeM8739, 2023-08-29:
Ahh ok. Yea i have an audio Jack wire but when i connected it - it wasn’t working. But then i also thought why should it be a recording issue with it since it’s connected to the receiver anyway and not the mic?? It could be my code so I’ll try adjusting the volume to what you recommended. But i can hear the greeting on my sd card great wired that way. The recording is the only issue for me.

PyrotekFX, 2023-08-29:
Yep I was the same, greeting came out super loud and clear, the recordings were barely audible, I only worked out it was a volume issue when I yelled at the phone out of frustration and it had the tiniest noise on the PC.

DianeM8739, 2023-08-29:
I’ll try that tonight when i work on it. Thanks!

PyrotekFX, 2023-08-29:
Not a problem, hope it works, did you work out the ticking issue though?

DianeM8739, 2023-08-29:
I got it!!! so i updated to what you recommended!! ok so i didnt need an audio jack. I connected (from the phone) REC - to LINE OUT L (audio shield) and + REC (phone) to LINE OUT G on the audio shield. no ticking sound!

PyrotekFX, 2023-08-30:
Got rid of the ticking noise too, Thank you for your help, hope all goes well with your jobs.

DianeM8739, 2023-08-30:
Thank you too!!!!

cp1797, 2023-10-07:
I have a Teensy 4.0 with Audio shield Rev D2 hooked up with soldered on headers. I've tried everything but the SD Card just doesn't detect. Keeps showing "Unable to access the SD card".

Going by [this comment](https://github.com/playfultechnology/audio-guestbook/issues/116#issuecomment-1703992567), I've checked with a multimeter the connection of the G pin between 5v and 3.3v and the GND pin and they are connected. CS is set to 10. I've also tried 7 through 14.

What am I doing wrong?? Would really appreciate the help.

DianeM8739, 2023-10-08:
So my issue was that i didn’t actually solder the pins together because i wanted to see if it works first. Once i soldered the pins to the audio shield it was able to recognize the sd card. Also make sure you reformat the sd card.

closed 2023-10-10 by alastaira