playfultechnology/audio-guestbook #119: Compiling Error

KaakaduRay, 2023-09-07:
Hi, Newbie to all
Trying to put the audio Guest book together but cannot get past compiling errors, help appreciated.
Arduino: 1.8.19 (Windows 10), TD: 1.58, Board: "Teensy 4.0, Serial + MTP Disk (Experimental), 150 MHz, Faster, US English, Bigger blocks (256 samples

    | 

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy\src/MTP_Storage.h:35,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy\src/MTP_Teensy.h:45,

                 from C:\Users\raylj\Documents\Arduino\audio-guestbook\audio-guestbook.ino:30:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:75:17: error: reference to 'File' is ambiguous

   75 |         virtual File openNextFile(uint8_t mode=0) = 0;

      |                 ^~~~

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/play_sd_raw.h:32,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/Audio.h:127,

                 from C:\Users\raylj\Documents\Arduino\audio-guestbook\audio-guestbook.ino:25:

C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:28:9: note: candidates are: 'class SDLib::File'

   28 |   class File : public Stream {

      |         ^~~~

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy\src/MTP_Storage.h:35,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy\src/MTP_Teensy.h:45,

                 from C:\Users\raylj\Documents\Arduino\audio-guestbook\audio-guestbook.ino:30:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:47:7: note:                 'class File'

   47 | class File;

      |       ^~~~

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h: In member function 'File File::openNextFile(uint8_t)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:216:33: error: 'class FileImpl' has no member named 'openNextFile'

  216 |                 return (f) ? f->openNextFile(mode) : *this;

      |                                 ^~~~~~~~~~~~

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h: At global scope:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:276:17: error: reference to 'File' is ambiguous

  276 |         virtual File open(const char *filename, uint8_t mode = FILE_READ) = 0;

      |                 ^~~~

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/play_sd_raw.h:32,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio/Audio.h:127,

                 from C:\Users\raylj\Documents\Arduino\audio-guestbook\audio-guestbook.ino:25:

C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:28:9: note: candidates are: 'class SDLib::File'

   28 |   class File : public Stream {

      |         ^~~~

[... 442 lines cut ...]
C:\Users\raylj\Documents\Arduino\audio-guestbook\audio-guestbook.ino:470:3: error: 'frec' was not declared in this scope; did you mean 'free'?

  470 |   frec.seek(0);

      |   ^~~~

      |   free

Using library Bounce in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Bounce (legacy)

Using library Audio at version 1.3 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio 

Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI 

Using library SD at version 1.2.4 in folder: C:\Program Files (x86)\Arduino\libraries\SD 

Using library SerialFlash at version 0.5 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SerialFlash 

Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 

Using library Time at version 1.6.1 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Time 

Using library MTP_Teensy at version 1.0.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy 

Error compiling for board Teensy 4.0.

alastaira, 2023-09-07:
The first relevant error message is "error: reference to 'File' is ambiguous"

That's because "File" is defined in the following two libraries:
C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:28:9: note: candidates are: 'class SDLib::File'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:47:7: note: 'class File'

Did you follow the steps about setting the default Teensyduino library location?

KaakaduRay, 2023-09-07:
Hi Alastaira
 Thankyou for responding much appreciated,
I believe I have the followed the install instructions correctly ? attached screen shots of file locations.

Regards
Ray 

From: alastaira
Sent: Thursday, 7 September 2023 11:58 PM
To: playfultechnology/audio-guestbook
Cc: KaakaduRay; Author
Subject: Re: [playfultechnology/audio-guestbook] Compiling Error (Issue #119)

The first relevant error message is "error: reference to 'File' is ambiguous"
That's because "File" is defined in the following two libraries:
C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:28:9: note: candidates are: 'class SDLib::File'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:47:7: note: 'class File'
Did you follow the steps about setting the default Teensyduino library location?

KaakaduRay, 2023-09-09:
Cleared all old files and reloaded sketch now compiles, however now some new issues to work through of  message recording and play back

closed 2023-09-09 by KaakaduRay