playfultechnology/audio-guestbook #124: No tone, or message. Recording but no sound recorded. Already looked through previous 'Issues' threads

256colorstoo, 2023-09-19:
Hi y'all! Just want to say that I really love how much help these threads have provided. 

I am running through:
-Teensy 4.0
-Rev D2

I'm having some trouble with audio that I know others have had. From the title, I am not hearing any sound from the speaker and nothing is recording from my microphone. There are files that have no sound, I increased the volume and still no hints of any audio. There is a file and is the length that I have the telephone off the plunger. So here's what I tried :

-Using the serial monitor, I can see that the code is working. When I push down the plunger I can see the code react to it.
I know that the speaker works because I plugged in my cellphone (Playing a youtube video) through the 3.5mm jack and I could hear the sound coming through the speaker.

-I have had troubles with a "unable to access SD card" until I resoldered some pins. And Then I was able to get it to work. My solders look good now and everything is connected properly. (For those who are stuck there. make sure that all the numbered pins of your teensy are connected to all numbered pins to your board)

-I've followed the wires from the speaker/microphone onto the phone board. I have screw on terminals, so i just hooked the bare wire around and tightened it. (Again I can prove this works because of the audio through the video playing off my cellphone). 

-I've used pin head connectors to connect the two boards. Then soldered the microphone to the board. And I've stripped wire to fit into the female connector pins. So everything is connected pretty well.

-my greeting is labeled "greeting.wav". at 44.1 khz and 16-bit. The "empty" audio files come back as .wav files. So ideally everything looks to be working fine.

In tools I have selected:
Board=teensy 4.0
usb type=Serial + MTP Disk (Experimental)
CPU speed=150 mhz
Optimize=Faster
Audio Tweaks=Bigger blocks (256 Samples)

My working theory:

I think it has to deal with the code. I did have to back track a lot to add in the "local.board" and other files. I'm thinking theres this disconnect from the code to the board, since the code is recording but not using the microphone as the input. I'm very much at a loss and I'm continuing to look into the code to find anything that may help me identify if there is a missing input or maybe I didnt read an instruction and didn't change something. 

pls halp. Gracias and any advice would be very appreciated.

brian-v8, 2023-09-21:
I am having this exact same issue. I need the phone by this Saturday so any help would be appreciated! I will update if I find a solution.

brian-v8, 2023-09-22:
After much troubleshooting, I tried this with another teensy processor with pre-soldered pins and everything works as it should. I am not sure if this was due to a faulty soldering job on my part or if the processor was bad. Double-check your solder joints and wiring.

256colorstoo, 2023-09-30:
Thank you @brian-v8 !!! So for anyone reading this he was right. I thought I had every pin connected but, due to some awkward placing of female connector boards, I had not connected every pin to the two boards. basically i had gaps and some of the points on the bottom board (9,17) weren't connected to the top board. 

In the future, CONNECT EVERY NUMBER FROM THE BOTTOM BOARD TO THE TOP BOARD. Now I have this awful jerry rigged mess that I may just buy the presoldered board for and star all over again.  Good news is that it works now. I have the wires from the phone (both receiver and speaker) connected straight to the board. Just make sure to find the plunger on the phone terminal, That need be connected still. 

Beyond that, I've just got to re-record my message and work out the volume issues ( which that isn't too much trouble) and just find a way to pack it all in. Either in the phone itself or an outside box like in the OG video. Thanks so much for the help, I now know way more about electronics than I did when I initially started and fundamentally look at problems differently. I've recently bought two (new to me) phones and I'm going to replicate this all over again... This time with KNOWLEDGE!!! This whole group has been a big help!!

closed 2023-09-30 by 256colorstoo